MicrosoftDocs/windows-driver-docs #1: What makes "Lid" Special?

PeterGV, 2016-08-01:
A small issue:

There are a significant number of "standard" devices shared between Windows and the ACPI BIOS (as noted in the first section on this page).  A few devices are called-out here for special treatment.  What makes them special?  Why is "Lid" listed here, and Battery, but not the Ambient Light Sensor device (for example)?

RE: https://github.com/Microsoft/windows-driver-docs/blob/master/windows-driver-docs-pr/bringup/acpi-defined-devices.md

tedhudek, 2016-08-02:
@barrygolden do you have any insight on this one?

barrygolden, 2016-08-02:
I’ll follow up with the ACPI PMs to see if we have additional specs or reference available on ambient light sensors.

PeterGV, 2016-08-02:
First… let me be entirely clear:  While this issue is a legitimate (if trivial) one, I entered it primarily as a way to test how issues filed on GitHub will be handled by the doc team, on behalf of the 3rd party driver development community.  So, my day-to-day business isn’t exactly impacted by the need to clarify this topic.
But… addressing the issue: To be clear… the question isn’t about “Ambient Light Sensors” per se.

The question is why do we specifically list Lid, among all the other possible devices that are specified in ACPI in this topic?  I chose Ambient Light Sensor as a random example.  I could have chosen the Power or Sleep buttons.

I guess the net thrust of my comment is that I find it odd that some specific ACPI devices of no particular distinction, but not others, would be listed in the WDK.  In essence, my question comes down to: What is this section “ACPI-defined devices” trying to accomplish?  Why is it a topic?  And why is Lid one of the devices listed?

I see that the Battery device, and the time and alarm device have (what looks to me to be) Windows or MSFT-specific information listed about them.  So, is this section supposed to be “ACPI-defined Devices and Windows”?  Is the idea to list the specific interpretation or requirements that Windows places on various standard APCI devices?  If so, there’s a LOT more that belongs in this section.  But it’s not clear that’s what this section is supposed to be fore.

Please read what the topic says about the Lid device.  It says, essentially, nothing.  Not even what Windows uses Lid for, or that there’s a _Lid control method.

Hope that further clarifies the issue I’m raising (and why I’m raising it),

Peter

barrygolden, 2016-08-03:
Created VSO task:  
Task 8375831: ACPI: ACPI-defined devices (rewrite topic per MVP input)

tedhudek, 2016-08-08:
This is tracked internally by the above, so closing out this issue.  Thanks, Peter, for pointing this out.

closed 2016-08-08 by tedhudek