MicrosoftDocs/windows-driver-docs #40: Formatting error on ordered list nested in unordered list

rpavlik, 2016-12-01:
Seen here: https://msdn.microsoft.com/en-us/windows/hardware/drivers/install/accessing-device-properties

tedhudek, 2016-12-01:
Thanks again @rpavlik, this is another good find.  I've updated the page in question.  If you like, consider a pull request next time.  Whatever's easier...  Thanks again for caring about the driver docs!

closed 2016-12-01 by tedhudek

rpavlik, 2016-12-01:
No problem - I didn't know if this was a rendering tool issue or a source/input issue (nesting those two kinds of lists in Markdown is something I usually don't dare attempt). I was really just trying to use the docs :D and don't know the recommended dev workflow here for testing a fix (Looks like the hidden powershell script does do something, though something that takes a long time), so I figured it was better to at least report it than ignore it because I couldn't fix it right away.

tedhudek, 2016-12-01:
Excellent point, Ryan!  I was not thinking about the rendering issue, which does indeed come from an oddity on our side.  You might have noticed that GitHub handled those extra line breaks with no problem.  So you're absolutely right, no way for you to test our internal rendering engine :)

rpavlik, 2016-12-02:
(well, the build script kind of works, but the generated files... leave a lot to be desired.)

tedhudek, 2016-12-02:
Is there anything in particular you are trying to accomplish?  Do you just want the docs as local HTML?