MicrosoftDocs/windows-driver-docs #45: Contradiction in network interface provider documentation

I-gor-C, 2016-12-19:
[Registering as interface provider](https://msdn.microsoft.com/windows/hardware/drivers/network/registering-as-an-interface-provider) explicitly states that any network driver type can serve as an interface provider:
> ... _However, miniport drivers and filter drivers can also be interface providers_

On the other hand, [OID_GEN_ADMIN_STATUS](https://msdn.microsoft.com/en-us/library/windows/hardware/ff569437(v=vs.85).aspx) documentation states the opposite: 
> _Only NDIS network interface providers, and therefore **not miniport drivers or filter drivers**, must support this OID as an OID request_.

tedhudek, 2016-12-22:
Investigating...

jtippet, 2016-12-26:
The first quote means:  A driver (.sys file) that is an NDIS miniport driver can also register as an interface provider.  In that case, the .sys file is both a miniport and an interface provider.

The second quote means that `OID_GEN_ADMIN_STATUS` (and similar OIDs) are not part of the contract for miniport drivers.  Your `MiniportOidRequest` handler will never receive `OID_GEN_ADMIN_STATUS`.  However, if your .sys file hosts both a miniport driver and an interface provider, then the interface provider aspect of your code needs to implement `OID_GEN_ADMIN_STATUS`.

It's worth pointing out that interface providers are niche stuff.  I'm not aware of any 3rd-party drivers that register as interface providers.  You can build one if you want, but 99% of cases are handled by the built-in interface providers.  Interface providers (and the OIDs that belong exclusively to them, like OID_GEN_ADMIN_STATUS) are typically not interesting and won't solve most problems.

I-gor-C, 2016-12-26:
Thank you Jeffrey. I think it's worth adding this clarification to the documentation to avoid confusion.

One thing still isn't clear to me, though - what is the meaning on "interface" in this context? Is registering an interface is something like installing a new adapter which appears in the network connections (such as the loopback adapter)? I haven't found any guide about interaction with the registered interface.

jtippet, 2016-12-26:
An "interface" is meant to correspond to [RFC 2863](https://tools.ietf.org/html/rfc2863).

NDIS has a built-in interface provider that automatically provides interfaces for:

- Every miniport driver
- Every filter (LWF) module
- The loopback interface

Interfaces show up in APIs like GetIfTable2Ex.  If you write your own interface provider, you can add entries to the table returned by GetIfTable2Ex.  In most cases, applications only care about the interfaces that NDIS provides for miniports, though.  (Actually, I tend to see applications go through some effort to filter out non-miniport interfaces.)

I-gor-C, 2016-12-26:
So if I want to add a virtual network interface via an existing miniport driver, this is not the way to go, right?
Adding a dedicated IM driver seems like a lot of effort and overhead... Can I add the IM flag to the miniport registration attributes to achieve similar effect?

(Sorry if I go out of scope to the original question...)

jtippet, 2017-02-01:
The interfaces that come via an interface provider don't have a datapath, don't bind to TCPIP, and won't show up in many GUIs.  So while you can use them to add another interface to your miniport, you'll probably be disappointed with the limited capabilities of this secondary interface.

closed 2017-02-02 by tedhudek