MicrosoftDocs/windows-driver-docs #66: Clarification on the LSO data

I-gor-C, 2017-03-22:
[Accessing TCP/IP Offload NET_BUFFER_LIST Information](https://msdn.microsoft.com/en-us/windows/hardware/drivers/network/accessing-tcp-ip-offload-net-buffer-list-information) mentions **TcpHeaderOffset** field in the **NDIS_TCP_LARGE_SEND_OFFLOAD_NET_BUFFER_LIST_INFO** structure - this field identifies the location of the TCP header offset to avoid the need to perform packet parsing (and avoid the associated performance impact). It's not stated, however, whether this implies that the TCP header is located in contiguous memory from the beginning of the packet or it can potentially be in another MDL. Without LSO the L4 header can definitely be in the second MDL (I've seen this with DHCP packets, for example). 
Can we have a clear statement about this in the documentation? Otherwise, the usefullness of **TcpHeaderOffset** is pretty limited.

P.S. And in overall, it would be very helpful to know the OS logic behind the decisions to split/not split packet headers to different MDLs.

jtippet, 2017-03-22:
The statement is on [another page](https://msdn.microsoft.com/en-us/windows/hardware/drivers/network/sending-ethernet-frames):

> Do not split the MAC header of the transmit Ethernet frame across multiple MDLs. Treat the Virtual LAN (VLAN) (or Priority) flag, if present, as part of the MAC header. Therefore, this flag must be in the same MDL as the rest of the MAC header.

In short, only the L2 header is guaranteed to be contiguous.  I can see how this makes things complicated if you're doing LSO or even checksum offload.  We'll investigate what we can do to make headers contiguous more often.

Unfortunately, because NDIS 6 did not require protocols to keep the TCP header in the first MDL, we can't make that restriction stricter, without breaking compatibility with 3rd party protocol drivers.

P.S.  The OS doesn't really decide to split packets into multiple MDLs.  Rather, the OS merely lacks optimizations to coalesce packets into the same MDL in some cases.  This is because the OS is architected in layers, roughly like the OSI model.  In theory, each layer independently prepends its own headers.  In practice, we do have optimizations to coalesce some of them (e.g., the IP and TCP headers are allocated together), since it would be silly to send every packet with 7 fragments.

As a matter of fact, the upcoming "Creators Update" of Windows 10 has one additional optimization to reduce the typical number of fragments on WLAN NICs from 3 to 2.

I-gor-C, 2017-03-22:
Thanks for the clarification, Jeffrey.
That's very unfortunate...
Regarding LSO - I haven't seen L3/L4 headers being split to a different MDL ever if LSO is enabled. Can this be enforced (maybe implicitly) by some upper layer module for LSO scenario, such as TCP/IP driver?

BTW, why would keeping the TCP header in the first MDL be problematic for 3rd-party protocol drivers? This is a valid scenario that must be handled today as well. AFAIU, forcing the header to the first MDL will only make some code in 3rd-party drivers unreachable without any functional impact - isn't it so?

jtippet, 2017-03-22:
We definitely continue to improve to Microsoft's TCPIP driver, and there's still more room for more MDL-combining optimizations.  But even if our TCPIP driver could always perfectly allocate enough headroom for all possible types of packets -- that's still not good enough to make a guarantee to the NIC driver.  Consider for example, a 3rd party virtual switch driver (which looks like a MUX driver in the host).  This driver doesn't contain a TCPIP stack, but it does deliver packets from a guest VM which does contain a TCPIP stack.  We have limited ability to retroactively change the rules for that 3rd party MUX driver.

But -- that doesn't mean we've given up.  We are exploring what we can do in the next version of NDIS to make the datapath simpler and to make fragmentation less of a problem.

I-gor-C, 2017-11-07:
Hi,

I'm reopening this issue since it looks that there is a piece of documentation that somewhat contradicts Jeffrey's statements above regarding headers splitting. This is a quote from the [Offloading the Segmentation of Large TCP Packets](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/offloading-the-segmentation-of-large-tcp-packets) page:

> All packet headers (Ethernet, IP, TCP) will be in the first MDL of the packet. The headers will not be split across multiple MDLs.

jtippet@, can you please address this statement? Is it correct or a documentation error?

Again, thanks a lot for your help.

duncanmacmichael, 2017-11-07:
Hi @I-gor-C , sorry for the confusion. I edited your comment just to fix the @ symbol in your reference, just FYI. If it is a documentation error I'll be happy to take care of it once we get some clarification. Thanks!

jtippet, 2017-11-14:
That statement is correct *when the TCPIP stack enables LSO*.  So in general, you can't assume that IP headers are in the same MDL as Ethernet headers.  But if LSO is enabled, you can make that assumption.

I don't think my earlier reply was particularly good, since I didn't mention this exception.  Sorry about that.

duncanmacmichael, 2017-11-14:
Thanks, @jtippet. I'll update the LSO doc to make it clearer. Also, thanks again for paying such close attention, @I-gor-C. Always good to help clarify the docs.

closed 2017-11-14 by duncanmacmichael

I-gor-C, 2017-11-14:
Thanks! Since I'm  mostly interested in LSO from the miniport driver perspective, it's helpful to know that the header will be contiguous if packet OOB states that LSO is on.

duncanmacmichael, 2017-11-14:
Definitely. The change has been made and should be live within 24 hours. :)