MicrosoftDocs/windows-driver-docs #130: Pages for !ppm.* debugger extn. command are empty placeholders (and some pages are missing altogether)

metathinker, 2017-05-29:
The following documentation pages for Debugging Tools for Windows debugger extension commands look like merely placeholders (entire page content: "The !blahblah command"). They should be filled in:

- `!ppmidle`: [debugger/-ppmidle.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/-ppmidle.md)
- `!ppmperf`: [debugger/-ppmperf.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/-ppmperf.md)
- `!ppmstate`: [debugger/-ppmstate.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/-ppmstate.md)
- `!ppmperfpolicy`: [debugger/-ppmperfpolicy.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/-ppmperfpolicy.md)
- `!ppmidleaccounting`: [debugger/-ppmidleaccounting.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/-ppmidleaccounting.md)

The same is true for the corresponding pages on https://docs.microsoft.com and https://msdn.microsoft.com , as well as the built-in Debugging Tools help (debugger.chm). Example for `!ppmidle`:
- https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-ppmidle
- https://msdn.microsoft.com/en-us/library/windows/hardware/hh439534(v=vs.85).aspx

---------------------------------------------------------------------------------

Unfortunately I don't know what these commands do, so I can't submit a PR to fill in the doc pages. However, I did find that all these extension commands appear to be kernel-mode-only and come from `kdexts.dll`.

In fact, kdexts exports some other commands that have no documentation at all, such as `!ppm` and `!ppmsettings` below. Those pages should be created and completed:

```
C:\> dumpbin.exe -nologo -headers -exports "C:\Program Files (x86)\Windows Kits\10\Debuggers\x64\winxp\kdexts.dll"
[snip snip]
        214   D5 000CE230 ppm = ppm
        215   D6 000CD9F0 ppmcheck = ppmcheck
        216   D7 000CF2B0 ppmhelp = ppmhelp
        217   D8 000CC160 ppmidle = ppmidle
        218   D9 000CA960 ppmidleaccounting = ppmidleaccounting
        219   DA 000C8DD0 ppmidlepolicy = ppmidlepolicy
        220   DB 000CEA30 ppmlpscenarios = ppmlpscenarios
        221   DC 000CD850 ppmperf = ppmperf
        222   DD 000C8B20 ppmperfpolicy = ppmperfpolicy
        223   DE 000CF0D0 ppmsettings = ppmsettings
        224   DF 000CB670 ppmstate = ppmstate
```

Only `!ppmhelp` comes up with useful help, and even that is not very complete:

```
0: kd> !kdexts.ppmhelp
KDEXTS supports the following PPM commands. Not all commands are supported on
all OS releases. The tag after the command (V78BTR) specifies whether the command
is supported on Vista, Windows 7, Windows 8, Windows Blue, Windows Threshold,
or Windows Redstone.

  !platformidle           ---BTR - display platform idle states
  !platformidleaccounting ---BTR - display platform idle state accounting
  !ppmidle                V78BTR - display processor idle states
  !ppmidleaccounting      V78BTR - display processor idle state accounting
  !ppmidlepolicy          V7---- - display C-state policy
  !ppmlpscenarios         --8--- - display per-scenario policy overrides.
  !ppmsettings            ----TR - display currently active ppm settings for the processor.
```

tedhudek, 2017-05-30:
@DOMARS please take a look

DOMARS, 2017-05-30:
I will contact the dev team and see if we can expand the documentation for these topics. Thanks for pointing out that they are missing content.

DOMARS, 2017-06-09:
I added some interim updates to provide very basic information on the commands. I will investigate to see if we can add more information. Thanks for pointing this out this gap in our documentation.

https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-ppmidle
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-ppmstate
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-ppmperfpolicy
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-ppmperf
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-ppmlpscenarios
https://review.docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-ppmidlepolicy
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-ppmidleaccounting

DOMARS, 2017-06-09:
I added some interim updates to provide very basic information on the commands. I will investigate to see if we can add more information. Thanks for pointing this out this gap in our documentation.

DOMARS, 2017-06-15:
Thanks again for this input. I hope the basic command information that was added is helpful.

closed 2017-06-15 by DOMARS

metathinker, 2017-06-15:
Thanks, this should be good for the short term. Hope to see more info later on if and when you get the same from the product team.