MicrosoftDocs/windows-driver-docs #140: Documentation of || (System Syntax)

thomasw-mitutoyo-ctl, 2017-06-06:
The documentation of || (System Syntax) hardly explains how a new system is created or attached to. Even the topic "Debugging Multiple Targets" does not give an example.

One way of creating a new system is the `.opendump` command. Maybe this could be mentioned at least.

tedhudek, 2017-06-06:
@DOMARS please take a look

DOMARS, 2017-06-10:
Thanks for this input. I will take a look at the topics and do some testing with WinDbg multiple targets.

DOMARS, 2017-06-13:
I updated these topics to include more information.
- || (System Syntax) 
- ||s (Set Current System)
- Debugging Multiple Targets
As I worked on this, I see how not running the go command can lead to a confusing situation, so I worked to highlight that.  
I added a simple example to the "Debugging Multiple Targets".  
Thanks again for the input.

thomasw-mitutoyo-ctl, 2017-06-14:
Thanks for the updated documentation. I appreciate this new way of providing feedback via Github. I was not informed about the progress of the emails I sent to wsddocfb@microsoft.com. This way is so transparent.

tedhudek, 2017-06-14:
So glad! We really like it too, it's great to hear from everyone and know what people care about

DOMARS, 2017-06-14:
Closing issue, thanks again for the input.

closed 2017-06-14 by DOMARS