MicrosoftDocs/windows-driver-docs #156: Table in wdi-transmit-operations-and-offloads page needs formatting

SachinDKulkarni, 2017-06-16:
The table in the "**TX Encapsulation: Population of 802.11 TX Frame Headers**" section in the following link:
https://docs.microsoft.com/en-us/windows-hardware/drivers/network/wdi-transmit-operations-and-offloads

(at the bottom of the page) needs formatting. Pasting the table content here for reference:

_Field name Subfield name Target-implemented encryption mode Host-implemented FIPS mode Set by host Set by target Set by host Set by target Frame Control Protocol Version X X Frame Control Type X X Frame Control Subtype X X Frame Control To DS X X Frame Control From DS X X Frame Control More Fragments X X Frame Control Retry X X Frame Control Pwr Mgmt X X Frame Control More Data X X Frame Control Protected Frame X X Frame Control Order X X Duration/Id X X Address 1 X X Address 2 X X Address 3 X X Sequence Control Fragment Number X X Sequence Control Sequence Number X X Address 4 X X QoS Control Added/populated by target. Added/populated by target in the case of 11n QoS association. HT Control Added/populated by target. Added/populated by target._

tedhudek, 2017-06-16:
Thanks @SachinDKulkarni.  @duncanmacmichael can help with this one when he returns on Monday

duncanmacmichael, 2017-06-20:
Hi @SachinDKulkarni, I saw this yesterday and will get it taken care of as soon as I can. I'll let you know when the table is fixed. Thanks for your patience!

SachinDKulkarni, 2017-06-21:
Thanks for the ack @duncanmacmichael. Will wait for the fix.

duncanmacmichael, 2017-06-21:
No problem @SachinDKulkarni . The fix has been pushed out and should go live within 24 hours. Thanks for finding this for us.

closed 2017-06-21 by duncanmacmichael