MicrosoftDocs/windows-driver-docs #180: WDI: RX decap expectations unclear

chaitu-tk, 2017-07-11:
https://docs.microsoft.com/en-us/windows-hardware/drivers/network/wdi-receive-operations-and-offloads 

"Rx decap | Replace the 802.3 header with a generic 802.11 header."

In the Rx side we receive a 802.11 packet, decap means converting it to 802.3 header before submitting to the WDI. The above statement is confusing, the statement should be

a) if WDI is expecting 802.3: Replace the 802.11 header with a 802.3 header.
b) if WDI is expecting 802.11 header:  No decap is needed as the header is already filled by the peer.
Generic 802.11 header doesn't make sense in this case (it is applicable only in TX where WDI fills some and expects Target/TAL to fill the rest).

The expectation of WDI is not clear from the existing doc.

barrygolden, 2017-07-11:
Thanks for filing this issue @chaitu-tk, we're reviewing your suggested changes.

duncanmacmichael, 2017-07-11:
Hi @chaitu-tk , thanks for pointing out that ambiguity with WDI expectations. The doc has been clarified and the changes will go live soon.

chaitu-tk, 2017-07-13:
Thanks, i am able to see the updates, but my intention was to document WDI expectations, the new updates make a conditional statement. Can you please check with WDI team as to expectations on RX header and then document?

tedhudek, 2017-07-13:
@duncanmacmichael please investigate

duncanmacmichael, 2017-07-13:
Ah, sorry if I misunderstood what you were trying to say, @chaitu-tk . I'll contact the WDI team and get their advice on the issue.

chaitu-tk, 2017-07-13:
No probs @duncanmacmichael will await your response.

duncanmacmichael, 2017-07-19:
Hi @chaitu-tk , I talked to the WDI team who proposed using an updated description with the original note. What do you think of this?
"Rx Decap | Replace non-initial A-MSDU subframe headers with generic 802.11 headers. | Target/TAL | During A-MSDU deaggregation, the non-initial MSDUs of the A-MSDU need their 802.3 header replaced by a generic 802.11 header. |

chaitu-tk, 2017-07-19:
Ok so basically WDI always expects 802.11 frames for both MSDU and A-MSDU's. May be we can explicitly add that statement it makes things more clear.

chaitu-tk, 2017-07-19:
Also it would be good to document the 802.11 generic header fields to be filled, similar to TX.

duncanmacmichael, 2017-07-19:
Good callout, let me check! :-)

duncanmacmichael, 2017-07-19:
OK, after re-checking with the WDI team, I've appended the statement that WDI always expects 802.11 headers to the note to make it clearer. As for documenting the 802.11 generic header fields on the Rx page, we don't have a specific table for that right now but I did amend the description to that effect as a solution for the time being. Thanks for your observations in improving the docs! The changes should be live within 24 hours.

closed 2017-07-19 by duncanmacmichael

chaitu-tk, 2017-07-21:
Thanks Duncan, i see the update.