MicrosoftDocs/windows-driver-docs #190: Convert \_ to _ in title and description metadata

nitoyon, 2017-07-21:
There are many `\_` in title and description metadata. They should be `_`.

### Example1: both title and description are broken

source code: [windows-driver-docs-pr/acpi/sending-an-ioctl-acpi-enum-children-request.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/2a2c6f25fa09542c04f48f3c9920ce67c7c8a812/windows-driver-docs-pr/acpi/sending-an-ioctl-acpi-enum-children-request.md)

```
---
title: Sending an IOCTL\_ACPI\_ENUM\_CHILDREN Request
author: windows-driver-content
description: Sending an IOCTL\_ACPI\_ENUM\_CHILDREN Request
```

Result HTML at https://docs.microsoft.com/en-us/windows-hardware/drivers/acpi/sending-an-ioctl-acpi-enum-children-request

```html
<head>
	<meta charset="utf-8" />
	<meta name="viewport" content="width=device-width, initial-scale=1.0" />
	<meta property="og:title" content="Sending an IOCTL\_ACPI\_ENUM\_CHILDREN Request" />
  :
<meta name="description" content="Sending an IOCTL\_ACPI\_ENUM\_CHILDREN Request" />
  :
	<title>Sending an IOCTL\_ACPI\_ENUM\_CHILDREN Request | Microsoft Docs</title>

```

It should be...

```html
<head>
	<meta charset="utf-8" />
	<meta name="viewport" content="width=device-width, initial-scale=1.0" />
	<meta property="og:title" content="Sending an IOCTL_ACPI_ENUM_CHILDREN Request" />
  :
<meta name="description" content="Sending an IOCTL_ACPI_ENUM_CHILDREN Request" />
  :
	<title>Sending an IOCTL_ACPI_ENUM_CHILDREN Request | Microsoft Docs</title>

```

### Example2: only description is broken

source code: [windows-driver-docs-pr/acpi/acpi-method-set-argument-integer.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/2a2c6f25fa09542c04f48f3c9920ce67c7c8a812/windows-driver-docs-pr/acpi/acpi-method-set-argument-integer.md)

```
---
title: ACPI_METHOD_SET_ARGUMENT_INTEGER macro
author: windows-driver-content
description: The ACPI\_METHOD\_SET\_ARGUMENT\_INTEGER macro sets the members of an ACPI\_METHOD\_ARGUMENT structure for a single integer value.
```

### Please fix it

At first I thought about fixing it myself, but I gave up because there are too many files to fix.

```
> git grep "title:.*_" | grep \\_
windows-driver-docs-pr/acpi/sending-an-ioctl-acpi-enum-children-request.md:title: Sending an IOCTL\_ACPI\_ENUM\_CHILDREN Request
windows-driver-docs-pr/audio/dsspeaker-directout-speaker-configuration.md:title: DSSPEAKER\_DIRECTOUT Speaker Configuration
windows-driver-docs-pr/audio/dsspeaker-surround-speaker-configuration.md:title: DSSPEAKER\_SURROUND Speaker Configuration
windows-driver-docs-pr/audio/obtaining-an-hdaudio-bus-interface-bdl-ddi-object.md:title: Obtaining an HDAUDIO\_BUS\_INTERFACE\_BDL DDI Object
windows-driver-docs-pr/audio/obtaining-an-hdaudio-bus-interface-ddi-object.md:title: Obtaining an HDAUDIO\_BUS\_INTERFACE DDI Object
windows-driver-docs-pr/audio/obtaining-an-hdaudio-bus-interface-v2-ddi-object.md:title: Obtaining an HDAUDIO\_BUS\_INTERFACE\_V2 DDI Object
windows-driver-docs-pr/bringup/acpi-device-specific-methods.md:title: Device-specific methods (\_DSM)
windows-driver-docs-pr/bringup/efi-battery-charging-completion-token.md:title: EFI\_BATTERY\_CHARGING\_COMPLETION\_TOKEN
windows-driver-docs-pr/bringup/efi-battery-charging-protocol.md:title: EFI\_BATTERY\_CHARGING\_PROTOCOL
windows-driver-docs-pr/bringup/efi-battery-charging-protocolchargebattery.md:title: EFI\_BATTERY\_CHARGING\_PROTOCOL.ChargeBattery
windows-driver-docs-pr/bringup/efi-battery-charging-protocolgetbatteryinformation.md:title: EFI\_BATTERY\_CHARGING\_PROTOCOL.GetBatteryInformation
windows-driver-docs-pr/bringup/efi-battery-charging-protocolgetbatterystatus.md:title: EFI\_BATTERY\_CHARGING\_PROTOCOL.GetBatteryStatus
windows-driver-docs-pr/bringup/efi-battery-charging-status.md:title: EFI\_BATTERY\_CHARGING\_STATUS
    :

> git grep "title:.*_" | grep \\_ | wc -l
1217

> git ls-files | wc -l
14142
```

We have to fix almost 10% of files!!!

tedhudek, 2017-07-21:
Thank for bringing this to our attention @nitoyon, I will investigate.

tedhudek, 2017-07-21:
Hi @nitoyon thanks again for opening this issue.  I just fixed it in cbf172e1ecc315d03d067cde8eaba16f5b604c46.

closed 2017-07-21 by tedhudek

nitoyon, 2017-07-22:
Thank you for your quick fix!!!