MicrosoftDocs/windows-driver-docs #221: Checksum offload disabling for a specific received packet

I-gor-C, 2017-09-06:
https://msdn.microsoft.com/library/windows/hardware/ff567877 states the following:

_Before indicating up a receive TCP/IP packet on which it performs checksum tasks, a miniport driver sets the appropriate XxxChecksumFailed or XxxChecksumSucceeded flags in the NDIS_TCP_IP_CHECKSUM_NET_BUFFER_LIST_INFO structure._

Does it mean that the miniport driver must perform checksum task on every received packet if the feature is declared as supported on init? What will happen if the miniport doesn't set neither XxxChecksumFailed nor XxxChecksumSucceeded? Will the OS validate the checksum in this case or it will treat the packet as invalid?

duncanmacmichael, 2017-09-07:
@jtippet Would you be able to elaborate? Thanks!

jtippet, 2017-09-08:
Checksum offload is disabled until the miniport receives OID_OFFLOAD_ENCAPSULATION.  At that point, the miniport is permitted to start validating the checksums on some received packets.  The miniport is not required to validate the checksum on every packet; if both the `ChecksumFailed` and `ChecksumSucceeded` flags are clear, then the OS will fall back to validating the checksum in software.

I-gor-C, 2017-09-08:
That clarifies it! Would it be possible to update the documentation?

duncanmacmichael, 2017-09-08:
Definitely. I can update the documentation for that and have it published next week. Thanks, @jtippet!

closed 2017-09-08 by duncanmacmichael