MicrosoftDocs/windows-driver-docs #254: Contradiction in *JumboPacket registry key definition

I-gor-C, 2017-10-17:
There are two contradicting definitions of *JumboPacket registry key in different parts of the documentation:

1.  Defined as Int with minimal and default values equal to 1514 and IHV defined maximum - https://docs.microsoft.com/en-us/windows-hardware/drivers/network/keywords-that-can-be-edited

2. Defined as Enum with supported values being 1514, 4088, and 9014 - subsection Device.Network.LAN.MTUSize in https://docs.microsoft.com/en-us/windows-hardware/design/compatibility/device-network-lan#device.network.lan.largesendoffload

duncanmacmichael, 2017-10-17:
Hi @I-gor-C , looking into this now. Thanks for pointing out the discrepancy.

duncanmacmichael, 2017-10-18:
Hey @I-gor-C, thanks for waiting. I worked with one of the NDIS developers to get clarification on this and submitted a PR on the latter page to fix the wording there. The "Keywords That Can Be Edited" page is technically correct in how to define the *JumboPackets keyword, so we changed the wording on the "Device.Network.LAN" page to explain how that keyword is defined by miniport drivers and how the drivers report their LSO max offload size to the OS at runtime. The PR should go live on that page within 24 hours. Thanks @Ray-Cai for your help on this.

I-gor-C, 2017-10-24:
Still unchanged...

duncanmacmichael, 2017-10-24:
Hmm, sorry about that. I'll see what's happening with the PR I submitted.

duncanmacmichael, 2017-10-25:
Thanks again for your patience, @I-gor-C . I've followed up and the PR on the Device.Network.LAN page has been completed. Look for the changes soon. :)

I-gor-C, 2017-10-25:
Hi Duncan, thanks for your efforts, I see the new wording now. However, it's somewhat puzzling now:

> The “*JumboPacket” standardized keyword in the Windows Registry is currently used for determining the MTU size for the LSO feature.
...
>When reporting the miniport’s LSO capabilities to the OS at runtime, miniport drivers use the MaxOffloadSize member of the NDIS_TCP_LARGE_SEND_OFFLOAD_V2 structure to reflect the "*JumboPacket" value.

Not sure I understand the relation to LSO in this statement. Jumbo packet is a network-level definition for large MTUs, while LSO allows to locally transfer an aggregation of multiple MTU-sized segments from the host to the NIC  - the data must still be chopped to MTU-sized buffers before physical transmission. Yet from the quote above it seems that maximum offload size for LSO is equal to the value of *JumboPacket. Is this the actual meaning or do I misunderstand something here? Maybe the MaxOffloadSize should be a multiple of *JumboPacket?

duncanmacmichael, 2017-10-26:
Ah, ok. I see what you're saying. I think we misunderstood your question and thought you were asking in relation to LSO because the Device.Network.LAN.MTUSize section is a sub-section under the Device.Network.LAN.LargeSendOffload section. Since you're just asking whether *JumboPacket should be an Int, as defined on the first page, or an enum, as defined on the second page, we'll look into that for you now and get back to you soon. Again, I appreciate your patience. :)

duncanmacmichael, 2017-10-27:
Hey @I-gor-C, so I did some more looking into this and came to the realization that both pages could be correct, but the "Keywords That Can Be Edited" page could be clarified a bit more. I believe *JumboPacket IS an enum, but its particular value is still an int because technically speaking enum values _are_ integers. Further up on the page, it is described that the "Int" type in the table simply means the value is numeric.

To confirm this, I used the "Get-NetAdapterAdvancedProperty" PowerShell cmdlet on a PC with an Ethernet adapter to see the values I could change, then tried to use "Set-NetAdapterAdvancedProperty" and change JumboPacket to a garbage value. In the error message it did tell me that supported values were 1514, 4088, and 9014, just like the "Device.Network.LAN" page said.

Therefore, I propose adding this wording to the "Keywords That Can Be Edited" page: "_*JumboPacket_ is an integer value, the possible values and maximum for which are IHV-defined in an enumeration."

Does that work? @Ray-Cai could you confirm this is correct?

I-gor-C, 2017-10-27:
Hi Duncan, thanks again.
The *JumboPacket registry is defined explicitly by each vendor. So in case your vendor has defined it in inf file to accept only 1514, 4088, and 9014 values (Intel NIC, I presume?), that's how it will be and different settings will be rejected by the registry service itself. Isn't it so? Another vendor might define it differently and in that case Set-NetAdapterAdvancedProperty might succeed. Please correct me if I'm off here.

In general, I'm more interested in the values that the vendor **_should_** define for that key, i.e. what is the Microsoft definition for valid values (even if they're not formally enforced). Maybe even both enum and integer are ok, but it should be stated consistently across the documentation.

duncanmacmichael, 2017-10-27:
Sure thing. You are correct in that it is an Intel NIC. :) 

After asking more questions, I have this for you. Basically, *JumboPacket can be either an int or an enum. The Windows code doesn't care. Some vendors let you set any int value between the minimum and the maximum, while others provide an enum list of possible values (like Intel in my example). I believe JumboPacket is defined as an Int (numeric) value on the Keywords page for instances like the former, where you could enter any number in the UI as long as it's between the min and the max.

Essentially, how it's defined in the INF doesn't matter; all that happens is the OS reads the values from the registry to set the MTU size. If it's a single int, then that's the MTU size if Jumbo Frames are enabled. If it's one member of an enum, whichever one is chosen is the MTU size (and again individual enum values are ints).

So, what would help clarify this here? I can say this for the JumboPacket description:

"The size, in bytes, of the largest supported Jumbo Packet (an Ethernet frame that is greater than 1514 bytes) that the hardware can support. This is also known as a Jumbo Frame. 

_*JumboPacket's_ range of values and maximum value are IHV-defined. Some vendors permit any value between their defined minimum and maximum, while others define an enumeration of supported values. For more info, check with your IHV."

We can massage that phrasing, but that's the gist.

I-gor-C, 2017-10-27:
Indeed, that's the answer I needed. I think your phrasing can be put as is in the documentations. And regarding the compatibility page, what's the process of opening a new issue about it?

duncanmacmichael, 2017-10-28:
Awesome, glad I could help. Your patience and tenacity are great. I just merged the edit into the master branch, so it should go live soon.

As for the other page, I would recommend opening up a new issue just like you did here on the repo level. The repo you're looking for is MicrosoftDocs/commercialization-public, as the page is located under the design section. I let the page owner know that her page might have to be updated, too, but it helps tremendously to have direct customer feedback. :) Like I said, pages under the Compatibility Program require more steps on the back end to edit and approve new content (to put it simply), so she can get the ball rolling with her contacts to make it happen.

Feel free to point back to this thread if anyone needs to reference it. Thanks again.

closed 2017-10-28 by duncanmacmichael