MicrosoftDocs/windows-driver-docs #260: Valid input for KeLowerIrql

I-gor-C, 2017-10-24:
KeLowerIrql documentation (https://msdn.microsoft.com/en-us/library/windows/hardware/ff552968(v=vs.85).aspx) currently describes the input parameter in this way:

NewIrql [in]
Specifies the IRQL that was returned from [KeRaiseIrql](https://msdn.microsoft.com/en-us/library/windows/hardware/ff553079(v=vs.85).aspx)

While it actually should be: 

NewIrql [in]
Specifies the IRQL that was returned from [KeRaiseIrql](https://msdn.microsoft.com/en-us/library/windows/hardware/ff553079(v=vs.85).aspx)/[KeRaiseIrqlToDpcLevel](https://msdn.microsoft.com/en-us/library/windows/hardware/ff553084(v=vs.85).aspx)

tedhudek, 2017-10-24:
Thanks @I-gor-C there might be a slight delay on this one because the owner is on vacation until next week

tedhudek, 2017-11-28:
Thanks for the note, @I-gor-C.  I've actually made the change to the doc source.  We're working on some internal publishing improvements right now that are blocking immediate republishing of the page, however.  Hoping to push it live by the end of the year.

closed 2017-11-28 by tedhudek