MicrosoftDocs/windows-driver-docs #277: Raw markdown on "Update a code sigining certificate"

ChrisMaddock, 2017-11-14:
Some raw markdown has slipped through half way down the page. Sorry, I'd submit a PR to fix myself, but I'm not familiar with what the formatting should be.

https://docs.microsoft.com/en-gb/windows-hardware/drivers/dashboard/update-a-code-signing-certificate
[image]

EliotSeattle, 2017-11-14:
Thanks for bringing this to our attention, Chris.

I made the fix and committed it to master. It will go live later today.

Eliot

https://docs.microsoft.com/en-gb/windows-hardware/drivers/dashboard/update-a-code-signing-certificate<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.microsoft.com%2Fen-gb%2Fwindows-hardware%2Fdrivers%2Fdashboard%2Fupdate-a-code-signing-certificate&data=02%7C01%7CEliot.Graff%40microsoft.com%7Cd1490c575098437c1d8c08d52b7589e8%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636462706184800860&sdata=JpT%2FUOo3Ly%2FEYC1R5gJbia7aMKkEyxa%2BBRbH%2B1NsBq8%3D&reserved=0>
[image]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F4837132%2F32788593-8eab1b2e-c951-11e7-842e-27b3479d0052.png&data=02%7C01%7CEliot.Graff%40microsoft.com%7Cd1490c575098437c1d8c08d52b7589e8%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636462706184800860&sdata=li1fws%2BQb9rlphe7vwC7cDp5fFAyqwt9HEF8Kx6XuXs%3D&reserved=0>

ChrisMaddock, 2017-11-14:
Great - thanks for the speedy fix!

closed 2017-11-15 by ChrisMaddock