MicrosoftDocs/windows-driver-docs #281: Unclear NDIS offload encapsulation configuration

I-gor-C, 2017-11-16:
There several items not well-defined in the description of [NDIS offload encapsulation](https://msdn.microsoft.com/library/windows/hardware/ff566702) (additional documentation here - https://docs.microsoft.com/en-us/windows-hardware/drivers/network/oid-offload-encapsulation): 

> A ULONG value that enables IPv4 encapsulation. A protocol driver sets Enabled to NDIS_OFFLOAD_SET_ON if it is enabling IPv4 large send offload version 1 (LSOV1), large send offload version 2 (LSOV2), or checksum offloads. Otherwise, the protocol driver sets Enabled to NDIS_OFFLOAD_SET_NO_CHANGE.

When receiving this OID in the underlying miniport driver, I see that in addition to NDIS_OFFLOAD_SET_ON and NDIS_OFFLOAD_SET_NO_CHANGE values there is also a third option - 0 (I assume it's the NDIS_OFFLOAD_SET_OFF in NDIS headers).

Besides that, it's not really clear how the miniport should handle NDIS_OFFLOAD_SET_ON and NDIS_OFFLOAD_SET_OFF requests. Should it turn on all the supported offloads for NDIS_OFFLOAD_SET_ON and turn all of them off for NDIS_OFFLOAD_SET_OFF? Does it relate to both Tx and Rx operations? I see a somewhat weird behavior in this aspect - whenever I disable LSO support in driver properties (but all the other offloads are on), I get NDIS_OFFLOAD_SET_OFF in the arriving OID_OFFLOAD_ENCAPSULATION.

Would really appreciate some clarification on the above.

duncanmacmichael, 2017-11-16:
Hi @I-gor-C, thanks for the question. Let me see if I can dig into this for you a bit and find an answer. I can then update one or both of the pages, depending on which one should be clarified.

duncanmacmichael, 2017-11-22:
Hey Igor, just letting you know I'm still working on this for you. Sorry, the dev who is the best candidate to answer this has been pretty busy.

duncanmacmichael, 2017-11-28:
Still getting an answer for you, thanks for your patience. I talked to him yesterday and it's on his list.

duncanmacmichael, 2017-12-01:
Alright @I-gor-C , I got an answer from the dev. He was quite detailed, so I'll give you his summary. First, the enums are defined as:

0 – NDIS_OFFLOAD_SET_NO_CHANGE
1 – NDIS_OFFLOAD_SET_ON
2 – NDIS_OFFLOAD_SET_OFF

So, he wanted to check if your "weird behavior" was a numerical value of 0 rather than 2? He thinks the OID_OFFLOAD_ENCAPSULATION sent to your miniport when turning LSO off was probably a NO_CHANGE instead of an OFF. Just checking since you had guessed that 2 meant OFF.

Here's his summary of the task offload OIDs and how to use them:

"OID_TCP_OFFLOAD_PARAMETERS is the one configuring each individual offloads (for some offloads, HW may not necessarily start doing them right away) , OID_OFFLOAD_ENCAPSULATION is used to activate all the configured/enabled offloads, or deactivate all offloads (HW starts to do offlaods). The encapsulation type miniport should use (3 types) is split between two OIDs. For handling no change, nothing should change. I believe the purpose of no change is also used for OID_TCP_OFFLOAD_PARAMETERS where some offload can be turned off while others don’t change. In general, I feel like most of the TCP/IP task offloads can be configured and activated with 1st OID."

Does that help?

I-gor-C, 2017-12-01:
Thanks, Duncan. That definitely helps. Regarding the value of NDIS_OFFLOAD_SET_OFF - I'm 95% sure that I saw it as a result of LSO being turned off. Yes, it was equal to 2, I made a typo above. In any case, I will verify this in a few days to be entirely confident. 

The explanation of the OID_OFFLOAD_ENCAPSULATION is indeed very helpful and I'm glad it confirms my assumptions. I'd suggest to put it in the OID documentation page, along with the addition of NDIS_OFFLOAD_SET_OFF.

duncanmacmichael, 2017-12-01:
Great! I'll update the two OID pages to add some clarification on the differences. Since NDIS_OFFLOAD_ENCAPSULATION specifies that it sets the Enabled member to NDIS_OFFLOAD_SET_ON in the case of LSOv1, LSOv2, or checksum, and sets it to NDIS_OFFLOAD_NO_CHANGE for everything else, it would follow that NDIS_OFFLOAD_SET_OFF would be used when the OID is sent to disable everything. I think. Let me know what you find out and if we need to re-open this issue. Thanks again!

closed 2017-12-01 by duncanmacmichael

I-gor-C, 2017-12-03:
Hi,

I've rechecked and indeed there is a weird behavior observed:

Once I disable LSO offloads in the driver properties and reenable the driver, I get OID_OFFLOAD_ENCAPSULATION oid with NDIS_OFFLOAD_SET_OFF value (== 2), thus effectively disabling all the offloads, not just LSO. This seems like unexpected behavior to me.

fridalisa, 2017-12-03:
Good evening. I understand that what you find looks strange, it does it for us to, that I can promise you. I try to understand how he manage to succeed all of his actions (especially the most criminal ones) around the clock day after day. Except when I sleep every third night, I realize that after 5 months I have just scratched the surface.

I'm very keen on what exactly the action you reported me is about. Is there any possibility that you can explain in short terms in "common language" what has happened and what it means. Does it have to with programming, addresses, audio and video capture, information retrieval, use of any service, etc.?

It is important if there is something illegal he made or used? Obviously, it's illegal to do anything in our devices, names, addresses, and social security numbers via remote control without our permission. but just this action you have been aware of, is it ok or is it illegal in any way?

Could you see information that could help me forward in our police investigation? Perhaps some number, perhaps signs that could be linked to the perpetrator? And if you could confirm that the something indicates that some action is prohibited or purely illegal, there would have been information in the form of gold.
 I obviously replace you if you want to help me about this man, since I have no paid membership, etc. Because I have no help from no one besides daily updates from RIPE NCC, they gave me a membership and ID for free long time ago because he utilizes their side to the max to be able to do all the evil creations.

 He uses all pages of barbarism for criminal activity. I found him on your side a few days ago too. He is like a parasite in infinite many disguises. he uses arin, lapnic, apnic, afrinic .. yes everything. He is incredibly wily and wobbles like a snake under all manipulation, prefixes and so on.

And of course, what can I do about this situation today to solve it out?

Sincerly Frida

duncanmacmichael, 2017-12-05:
So what you're saying is that when you disable LSO and see that OID come to your driver with NDIS_OFFLOAD_SET_OFF, all the offloads are disabled? That is correct according to the explanation I was given:

"This OID is used to activate all configured or enabled offloads, or deactivate all offloads (in other words, the hardware starts to perform the offloads). It does not provide fine control over individual offloads. OID_TCP_OFFLOAD_PARAMETERS is used to configure individual offloads and can also activate them. Generally, most TCP/IP task offloads can be configured and activated with OID_TCP_OFFLOAD_PARAMETERS."

To dig a bit deeper, is the weird behavior that NDIS is sending _this_ OID in a response to the request to turn off one offload (LSO)? OID_OFFLOAD_ENCAPSULATION is indeed intended to either activate or deactivate all offloads as far as I understand, so what it's doing seems to be correct. But, you're saying that if you request one offload to be disabled then you should be seeing the other OID instead? 

Just trying to understand your situation. :)

I-gor-C, 2017-12-05:
Hi Duncan,

Yes, you're correct. I wouldn't expect to see this OID with 'off' value when I've requested to disable only the LSO offload. In fact, the LSO offload is already disabled upon init since the configuration is being read from the registry. Thus sending the encapsulation OID doesn't provide any added value and only harms the performance (unexpected by the user!). 

As a side note - it would be more reasonable to send the encapsulation OID with 'off' value in the opposite situation, i.e. if checksum offloads were disabled rather than LSO - because LSO offload functionality requires checksum offloads to be enabled as well.

duncanmacmichael, 2017-12-05:
Okay, good to know. Let me see what I can find out for you.

I-gor-C, 2018-01-17:
Hi Duncan,

Sorry to bring this up again, I see that https://msdn.microsoft.com/library/windows/hardware/ff566702 is still not updated (e.g., there is no mentioning of NDIS_OFFLOAD_SET_OFF).

duncanmacmichael, 2018-01-17:
Hey @I-gor-C, no worries. :) If anything, I should be the one saying sorry for letting this slip. I'll see about updating the page to mention NDIS_OFFLOAD_SET_OFF per our discussion.

duncanmacmichael, 2018-01-18:
Okay, how's this: 

A ULONG value that enables IPv4 encapsulation. A protocol driver sets Enabled to NDIS_OFFLOAD_SET_ON if it is enabling IPv4 large send offload version 1 (LSOV1), large send offload version 2 (LSOV2), or checksum offloads. If a protocol driver is deactivating all offloads, it sets Enabled to NDIS_OFFLOAD_SET_OFF. Otherwise, the protocol driver sets Enabled to NDIS_OFFLOAD_SET_NO_CHANGE.

I can get that change added in today if that works, then it will go live within the next few days.

I-gor-C, 2018-01-18:
Sounds good!