MicrosoftDocs/windows-driver-docs #301: Visual Studio Community Compatibility

jnrbo, 2017-12-28:
I'm trying to follow the sample tutorial, but It's deprecated - some menus and options dosn't exist . 

I'm using:

```
Microsoft Visual Studio Community 2017 
Version 15.5.2
VisualStudio.15.Release/15.5.2+27130.2010
Microsoft .NET Framework
Version 4.7.02558
```

tedhudek, 2017-12-29:
Hi @jnrbo what's the URL of the sample tutorial you were trying to follow, and is there a particular steps or set of steps where you found that the menus and options weren't available?

jnrbo, 2017-12-29:
Hi @tedhudek 

I'm following this [tutorial](https://docs.microsoft.com/en-us/windows-hardware/drivers/print/building-a-basic-v4-printer-driver).

All menus inside **Property Pages window** are deprecated!

Thank you!

tedhudek, 2017-12-29:
Thanks @jnrbo I've assigned this issue to @barrygolden, who is out for the holidays, but can address this when he returns.  Apologies for the discrepancies, that's frustrating when you're trying to follow a tutorial.

barrygolden, 2018-01-03:
Investigating issue with deprecated menus and options. Working with print driver devs to replicate.

barrygolden, 2018-02-07:
@MFenelon, can you take a look at this issue?

closed 2018-02-13 by barrygolden