MicrosoftDocs/windows-driver-docs #310: Outdated NDIS_OFFLOAD structure documentation

I-gor-C, 2018-01-17:
The [documentation](https://msdn.microsoft.com/en-us/library/windows/hardware/ff566599(v=vs.85).aspx) relates only to NDIS versions up to 6.3.

```
Set the Revision and Size members of the NDIS_OBJECT_HEADER structure as follows:
For NDIS 6.30 and later drivers:
- Set Revision to NDIS_OFFLOAD_REVISION_3 (NDIS 6.30).
- Set Size to NDIS_SIZEOF_NDIS_OFFLOAD_REVISION_3.
For NDIS 6.1 and 6.20 drivers:
- Set Revision to NDIS_OFFLOAD_REVISION_2 (NDIS 6.1).
- Set Size to NDIS_SIZEOF_NDIS_OFFLOAD_REVISION_2.
For NDIS 6.0 drivers:
- Set Revision to NDIS_OFFLOAD_REVISION_1 (NDIS 6.0).
- Set Size to NDIS_SIZEOF_NDIS_OFFLOAD_REVISION_1.
```

A paragraph for NDIS 6.5 and later should be added:
```
For NDIS 6.50 and later drivers:
- Set Revision to NDIS_OFFLOAD_REVISION_4 (NDIS 6.50).
- Set Size to NDIS_SIZEOF_NDIS_OFFLOAD_REVISION_4.
```

And the wording 

> For NDIS 6.30 and later drivers

 should be changed to 

> For NDIS 6.30 and 6.40 drivers

duncanmacmichael, 2018-01-17:
Hey @I-gor-C, thanks for the heads up. I'll work on getting this updated with the newest info available.

duncanmacmichael, 2018-01-18:
OK, I have an update for you my friend. I asked Jeffrey about these new members, and the other new one that got added in NDIS 6.70. He told me what they were (VXLAN for the 6.50 additions and 464XLAT for the 6.70 addition). Basically, VXLAN offload is the same as NVGRE offload, but with a different protocol. 464XLAT is a feature related to mobile broadband that is for converting between IPv4 and IPv6. They're both kind of niche features.

So, I will indeed document these. However, at the moment, my team is diving into documentation for new features in the upcoming Windows release, so that is my priority for now. I will revisit these for sure when that is done.

Does that help for now? I wasn't sure why you were asking - did you want to use this new member and saw that it wasn't documented, or did you just catch that it wasn't documented and were being helpful? :)

duncanmacmichael, 2018-01-18:
Yeah, absolutely. You're right - newer versions of the structure should be indeed be used in newer versions of Windows if possible. Thanks for bringing this to my attention! I'll let you know when I make the changes.

closed 2018-01-18 by duncanmacmichael