MicrosoftDocs/windows-driver-docs #316: about SUPL?

nanji2018, 2018-01-25:
Hello:

i am an engineer in spreadtrum,which is a chip company.
about gnss driver,i have some questions.

firstly,about supl,from your gnss-driver-architecture,supl is provided by IHV designed.
will supl  be designed in windows or our chip os?

thanks

nanji2018, 2018-01-29:
hello barrygolden:

now we want to use MBB style to design GNSS model,but we are not sure about supl.
should we realize supl in windows？because in android，supl exist in framework，so we want to know whether windows also support supl now？

thanks

barrygolden, 2018-02-07:
Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.
I assume you’ve looked through the two GNSS driver topics that discuss SUPL:
-   [GNSS driver architecture](https://docs.microsoft.com/en-us/windows-hardware/drivers/gnss/gnss-driver-architecture)
-   [GNSS driver requirements](https://docs.microsoft.com/en-us/windows-hardware/drivers/gnss/gnss-driver-requirements)
Also, you may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.
-   [MSDN Windows Hardware WDK and Driver Development forum](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
-   [NTDEV forum at OSR Online](https://www.osronline.com/showlists.cfm?list=ntdev)
Again, thanks for reaching out. Best of luck with this issue.

closed 2018-02-07 by barrygolden