MicrosoftDocs/windows-driver-docs #331: Driver Installation instructions says to "Create a new port" of type "3D Port" which doesn't exist

Keavon, 2018-02-11:
On https://docs.microsoft.com/en-us/windows-hardware/drivers/3dprint/driver-installation I have followed all the instructions up to step 10, at which point it says

> Choose **Create a new port** and select **3D Port** for the type of port, then click **Next**.

On this screen in Windows 10, I am only presented with the options "Local Port" and "Standard TCP/IP Port", but "3D Port" does not show up. How can I proceed with installing this?

barrygolden, 2018-02-12:
Investigating missing "3D Port" issue.

barrygolden, 2018-02-12:
@Keavon, have you installed the applicable generic 3D printer driver from the Microsoft Update Catalog?

http://www.catalog.update.microsoft.com/Search.aspx?q=USB%5CMS_COMP_3DPRINT

Keavon, 2018-02-13:
@barrygolden could you please clarify which one of those I should install, and how to install it? I followed [these instructions](https://winaero.com/blog/how-to-install-cab-and-msu-updates-in-windows-10/) and tried every `.cab` file from the download page you linked, but each gave this error:

```
Deployment Image Servicing and Management tool
Version: 10.0.16299.15

Image Version: 10.0.16299.192

An error occurred trying to open - C:\Users\Keavon\Desktop\20764851_327f3bed314c128a6c749d2a3acd1967a3477bca.cab Error: 0x80070002

Error: 2

The system cannot find the file specified.

The DISM log file can be found at C:\Windows\Logs\DISM\dism.log
```

I also tried extracting the `MS3DPrintUSB.inf` file from inside the `.cab` then right-clicking and selecting Install but this just pops up with "*Install Error* | The third-party INF does not contain digital signature information". Using this as the `.inf` in Step 4 of the original guide pops up with "*Select Device* | The folder you specified doesn't contain a compatible software driver for your device. If the folder contains a driver, make sure it is designed to work on Windows for x64-based systems."

The "3D Port" option is still missing. I am running the latest 64-bit Windows 10 Pro installed less than a week ago from an ISO I burned to a DVD from the Microsoft media creation tool.

barrygolden, 2018-02-13:
Thanks for providing the detailed description above. Unfortunately, the Windows Driver Developer Documentation team doesn't have the resources to offer specific individual technical advice to the broad array of customers who use the driver documentation.

You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

[MSDN Windows Hardware WDK and Driver Development forum](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
 [NTDEV forum at OSR Online](https://www.osronline.com/showlists.cfm?list=ntdev)

Be sure to include relevant information about your driver development environment:
- Visual Studio version 
- Windows SDK version
- Windows Driver Kit (WDK) version
- Microsoft 3D Printing SDK version (latest version is 1.3)

Thanks for reaching out. Best of luck with this issue.

closed 2018-02-13 by barrygolden