MicrosoftDocs/windows-driver-docs #335: Instructions for getting USBView with VS2017 are incorrect

ocdtrekkie, 2018-02-14:
This is a rats nest for getting a single simple EXE, and Microsoft should mostly be ashamed for how hard they have made this. (Seriously, why not publish it in the Microsoft Store?)

Fixing this docs circle could help, but there is so much more wrong here than just the Docs.

**Microsoft Docs chain of shame**

Anyways, there are several new Docs pages I went to as part of this, and all of them are unhelpful. This issue is a chain of fail, as I recreate the browsing path for you.

USBView: https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview
- Get it from Debugging Tools for Windows

Debugging Tools for Windows: https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/index
- Get it from Windows Driver Kit (inside Visual Studio 2017)
- Note that Windows Driver Kit is not an option in VS2017's installer

Windows Driver Kit actually links me here: https://developer.microsoft.com/en-us/windows/hardware/download-kits-windows-hardware-development which then annoyingly bounces me back into Docs here:
Windows Driver Kit: https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk

Which tells me to *install Desktop development with C++ role*, which allegedly includes the Windows SDK. Which I had already installed, and it didn't have USBView. I added the whole role anyways, with an alleged cost of 1.05 GB, and still don't have USBView.

Back back out to Debugging Tools for Windows, where another option said I could get the Windows SDK: https://developer.microsoft.com/en-us/windows/downloads/windows-10-sdk
- Provides a standalone EXE.
- This might actually work, but I already _have_ the Windows SDK in VS2017. What gives?

A third option of how to get Debugging Tools for Windows on the same page linked above literally just provides the exact same URL as the above Windows SDK option.

**In comparison**

FTDI has a link with a zip install here: http://www.ftdichip.com/Support/Utilities.htm#MicrosoftUSBView

I could've saved myself a half hour if I didn't bother trying to get the latest possible version by going through Microsoft.

**Secondary issue**

All of these articles say I can get USBView/the Debugging Tools for Windows/the Windows SDK via Visual Studio, absolutely none of them specify _where_ Visual Studio will put it. So while I've searched pretty thoroughly, and suspect it isn't installed... it's entirely possible it might be and the Docs just haven't provided me any help in finding it.

DOMARS, 2018-02-22:
Thanks for this input. I will work to make provide more detail in the directions on how to download and locate the USBVIEW Utility.

DOMARS, 2018-02-28:
I have provided more detail on how to download and locate USBView and updated related download links. Thanks for the input.

closed 2018-02-28 by DOMARS

ocdtrekkie, 2018-02-28:
Thank you! I definitely confirmed despite selecting the various roles I was told contained this in Visual Studio, I did not have the Debuggers folder until I installed the EXE download version. The document is now unambiguous on what the best way to get this is, and I've found it now!

Thanks!