MicrosoftDocs/windows-driver-docs #341: need more examples on filtering and displaying parameters

binarycrusader, 2018-02-28:
A common task when debugging multi-threaded issues is to get a particular field from a parameter value for every thread that has a specific function in its callstack; after hours of experimentation I finally managed to discover the syntax I needed for my particular case:

dx @$curprocess.Threads.Where(t => t.Stack.Frames.Any(f => f.ToDisplayString().Contains("::Finalize "))).Select(t => t.Stack.Frames.Single(f => f.ToDisplayString().Contains("::Finalize "))).Select(f => f.Parameters).Select(p => p.pPublicSession->m_pSession->m_sczClientString)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a06a3c75-fe34-acfa-51ef-4296795e8f70
* Version Independent ID: 5c3bdaf5-eec2-3ea2-3c5b-f53f7b80b41b
* [Content](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/using-linq-with-the-debugger-objects#supported-linq-syntax---query-methods)
* [Content Source](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/using-linq-with-the-debugger-objects.md)
* Service: unspecified

DOMARS, 2018-02-28:
Thanks for this input, I will take a look at this example, seems like it would be useful for others.

DOMARS, 2018-03-02:
Per our offline discussion, I will close this issue as we work to come up with a usable sample.

closed 2018-03-02 by DOMARS