MicrosoftDocs/windows-driver-docs #344: It would be nice to know how to deactivate remote debugging

efalk, 2018-03-01:
Maybe I just have to quit out of windbg, but if there's a command to "undo" the .server command, it would be nice to know about it.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1cfcfd39-550e-9eab-5873-ec677b2537eb
* Version Independent ID: 92f78106-388e-a61b-dd42-9bfe433f72e6
* [Content](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/activating-a-debugging-server)
* [Content Source](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/activating-a-debugging-server.md)
* Service: unspecified

DOMARS, 2018-03-02:
@efalk I added links to these topics. I believe those should be helpful. Those changes should be live shortly.
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/controlling-a-remote-debugging-session
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-endsrv--end-debugging-server-
Thanks for the input.

DOMARS, 2018-03-12:
The new links listed above are live, thanks for the input.

closed 2018-03-12 by DOMARS