MicrosoftDocs/windows-driver-docs #346: Empty article

sahikon, 2018-03-01:
In the past, it was part of API reserved for internal use.
https://msdn.microsoft.com/en-us/library/windows/hardware/ff549753(v=vs.85).aspx

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 88432141-902e-3fc5-64ed-3ee403aace53
* Version Independent ID: 73bb9eae-c60d-6615-9e31-9f4da4599caa
* [Content](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/cm-open-devnode-key-ex)
* [Content Source](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/cm-open-devnode-key-ex.md)
* Service: unspecified

tedhudek, 2018-03-12:
Hi @sahikon could you be more specific?  I'm not sure what change you are requesting to the documentation.  Thanks!

sahikon, 2018-03-12:
Hello @tedhudek ,

Let me clarify request.

My headline "Empty article" means that [this](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/cm-open-devnode-key-ex) is empty and does not provide any information about CM_Open_DevNode_Key_Ex:

[image]

CM_Open_DevNode_Key_Ex is marked as "private" at MSDN.

The question: if we want to keep a separate page for CM_Open_DevNode_Key_Ex (and other "private" APIs), then is there a need to add some sort of description to that page? E.g., if CM_Open_DevNode_Key_Ex is still considered as private, then I guess one sentence like "This function is  reserved for use by the Windows operating system" would be enough.

tedhudek, 2018-03-13:
Thanks @sahikon it's a good point. I've added a sentence to the pages, for example https://docs.microsoft.com/en-us/windows-hardware/drivers/install/cm-create-devnode.

closed 2018-03-13 by tedhudek