MicrosoftDocs/windows-driver-docs #347: what abot IoT Core?

dvescovi1, 2018-03-02:
talk about IoT Core requirements

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a1fe9907-f850-a8ae-112b-30d154468fe1
* Version Independent ID: b1f64595-9c3c-61ce-4c51-84c642371af4
* [Content](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/minimum-uefi-requirements-for-windows-on-soc-platforms)
* [Content Source](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/minimum-uefi-requirements-for-windows-on-soc-platforms.md)
* Service: unspecified

barrygolden, 2018-03-02:
Self-assigned @barrygolden for investigation.

barrygolden, 2018-03-02:
Initiated contact with IoT Core team resources for information on UEFI requirements for IoT Core.

barrygolden, 2018-07-25:
@dvescovi1, thanks for reaching out to the Windows driver developer documentation team and filing an issue on the docs regarding UEFI on IoT vs. traditional Windows platforms. We've filed an internal work item (18382061) to add more information about this in a future documentation update.

If you have specific scenarios that you'd like discuss with the IoT or driver developer community, we recommend that you visit the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

[MSDN Windows Hardware WDK and Driver Development forum](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

[NTDEV forum at OSR Online](https://www.osronline.com/showlists.cfm?list=ntdev)

Feel free to submit a pull request on the driver docs to help us benefit from your knowledge and experience with the WDK.

closed 2018-07-25 by barrygolden