MicrosoftDocs/windows-driver-docs #352: Add registry paths as well?

jithinjosepkl, 2018-03-06:
Really great info, and well written. Can you add the registry location for the dump type configuration as well?
(Same request for user-mode dump files).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 8a6515f5-f476-5c1f-5c9b-0f4bc489bfb3
* Version Independent ID: e875b4a7-3499-0ea1-0889-fbb0a5288ba8
* Content: [Active Memory Dump | Microsoft Docs](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/active-memory-dump)
* Content Source: [windows-driver-docs-pr/debugger/active-memory-dump.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/active-memory-dump.md)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-03-06:
Thanks, Jithin. The writer for the debugger content is out this week, but he'll address this when he returns.

DOMARS, 2018-03-13:
I'm checking with the dev team on this request.

DOMARS, 2018-08-06:
@jithinjosepkl - Additional information in registry keys for user mode dump are described here.
https://docs.microsoft.com/en-us/windows/desktop/wer/collecting-user-mode-dumps

We can follow up out of band to see if the dev team would like to document the reg keys for Kernel Mode.

closed 2018-08-06 by DOMARS