MicrosoftDocs/windows-driver-docs #361: Could you fix font error?

weignv, 2018-03-12:
Noticed that there are multiple arenâ€™t, which should be aren't. This error is shown in multiple web browsers and please take look at it.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 49099536-c363-e94d-ce10-44454344a8cd
* Version Independent ID: 4b143eee-1dd7-d523-dd2f-df506d592314
* Content: [Tile resources | Microsoft Docs](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/tile-resources)
* Content Source: [windows-driver-docs-pr/display/tile-resources.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/tile-resources.md)
* Service: **unspecified**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-03-12:
Thanks, @weignv. I got the instances in this topic. @tedhudek, can you scan for other occurrences and  log a bug to me?

EliotSeattle, 2018-03-19:
The fix is in a commit for 22 files. Will update when this goes live.

EliotSeattle, 2018-03-21:
These were merged to master. Thanks, @weignv, for pointing this out.

closed 2018-03-21 by EliotSeattle

weignv, 2018-03-21:
Thanks for fixing it.