MicrosoftDocs/windows-driver-docs #362: Missing entry for 0x41793

TruePikachu, 2018-03-14:
Parameter 1 for this bugcheck can be `0x41793`, but it is not listed in the table. Anyone know what it means?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT | Microsoft Docs](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-03-16:
In the next update cycle for the stop codes I will see if it is possible to include additional information, such as for the requested parameter 1 value.

Anonymous-W, 2018-04-07:
I met same problem too,looking forward to the information

DOMARS, 2018-06-25:
Planning to consolidate the debugging issues and get with the dev team this week.

DOMARS, 2018-08-17:
@HeinziAT - I'm following up with the dev team to see if we can document this. Thanks for the interest in improving our docs!

DOMARS, 2018-09-13:
@TruePikachu and @Anonymous-W - I worked with the dev team to add an entry for 41793. Thanks for the request and your interest in improving our docs!

closed 2018-09-13 by DOMARS