MicrosoftDocs/windows-driver-docs #365: OS version

sureshkumarct, 2018-03-14:
It would be better if you could add the OS version for Win 10 also

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0afbc04b-349a-57ed-eeb0-3d2cd28e359c
* Version Independent ID: de61f684-31d7-2de5-d66c-a3319f5198d9
* Content: [Inf2Cat | Microsoft Docs](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/inf2cat)
* Content Source: [windows-driver-docs-pr/devtest/inf2cat.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/inf2cat.md)
* Service: **unspecified**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-03-21:
This is fixed, thanks @sureshkumarct for the suggestion.

closed 2018-03-21 by tedhudek

nospamsam2011, 2019-02-24:
Why is inf2cat not included with the Win10 SDK?  If it is, where is it located?