MicrosoftDocs/windows-driver-docs #369: Docs incorrect on allowed hybrid drivers SDV works on

jcbottorff, 2018-03-14:
The docs say: The driver does not combine driver models (for example, KMDF with WDM, or KMDF and NDIS). SDV does work with two way hybrid drivers, but not three way hybrids.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b5fd6a7f-6aab-f9b0-e4ab-ea54383adc80
* Version Independent ID: b3bd9ec2-cc7e-98c6-b445-045f2ac4a481
* Content: [Determining if Static Driver Verifier supports your driver or library | Microsoft Docs](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/determining-if-static-driver-verifier-supports-your-driver-or-library)
* Content Source: [windows-driver-docs-pr/devtest/determining-if-static-driver-verifier-supports-your-driver-or-library.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/determining-if-static-driver-verifier-supports-your-driver-or-library.md)
* Service: **unspecified**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-03-19:
Don, can you take a look at this feedback for Static Driver Verifier and correct content if it's true?

EliotSeattle, 2018-03-20:
@aahill please verify this request and add it if appropriate. Thanks, E.

aahill, 2018-05-21:
Hi @jcbottorff , sorry for the delay on this; I've reached out to the SDV team, who plan to address this in an upcoming doc update. Thanks so much for letting us know about this.

closed 2018-05-21 by aahill