MicrosoftDocs/windows-driver-docs #372: Build Solution Error

DonotRight, 2018-03-19:
I add USB device capability to the app manifest like this article's sample
<Capabilities>
    <DeviceCapability Name="usb" />
    
    <Device Id = "vidpid:03F0 0C54">
      
      <Function Type ="name:vendorSpecific"/>
    </Device>
  </Capabilities>

then Build Solution, there is 1 Error comes out like below

Error		Validation error. error C00CE014: App manifest validation error: The app manifest must be valid as per schema: Line 34, Column 6, Reason: Element '{http://schemas.microsoft.com/appx/manifest/foundation/windows10}Device' is unexpected according to content model of parent element '{http://schemas.microsoft.com/appx/manifest/foundation/windows10}Capabilities'. Expecting: {http://schemas.microsoft.com/appx/manifest/foundation/windows10}DeviceCapability.

Why?		

---
#### 文件詳細資料

⚠ *請勿編輯此區段。 這是 docs.microsoft.com ➟ GitHub 問題連結的必要項。*

* ID: b843acc8-3fa9-5f3a-a89b-9c8149914020
* Version Independent ID: 1c3e8047-ad68-29a0-d3e7-9bf842c27885
* Content: [How to add USB device capabilities to the app manifest](https://docs.microsoft.com/zh-tw/windows-hardware/drivers/usbcon/updating-the-app-manifest-with-usb-device-capabilities)
* Content Source: [windows-driver-docs-pr/usbcon/updating-the-app-manifest-with-usb-device-capabilities.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/updating-the-app-manifest-with-usb-device-capabilities.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

PageWriter-MSFT, 2018-03-19:
Hello:

Thanks for reaching out. Please provide your app manifest as it is difficult to detect the error is without looking at it. 

Page Writer

DonotRight, 2018-03-21:
Thank for your reply.
I think it is my fault to direct copy the sample on this article to my Package.appxmanifest
like this
<Capabilities>
      
      <m2:DeviceCapability Name="usb">
          
          <m2:Device Id="vidpid:0547 1002">
              <m2:Function Type="classId:ff * *"/>
              
          </m2:Device>
          
          <m2:Device Id="vidpid:045E 0611">
              
              <m2:Function Type="name:vendorSpecific"/>
          </m2b:Device>
      </m2:DeviceCapability>
  </Capabilities>

under Win10 & Visual Studio 2017 UWP app project.
Then I try to kick off all "m2" betweent <Capabilities> and </Capabilities>,
after that I rebuild solution, no error pump up again, so I can continue again.
TKS! again.

PageWriter-MSFT, 2018-04-23:
Great! Let us know if you hit some documentation issues.

PageWriter

closed 2018-04-23 by PageWriter-MSFT