MicrosoftDocs/windows-driver-docs #376: Default sample rates for USB devices

endolith, 2018-03-19:
> When the SysAudio system driver initializes the KMixer driver for an audio device, it assigns a sample rate corresponding to the higher of 44.1 kHz and the highest rate available on the audio device.

This doesn't seem true for USB Audio devices, which default to 16-bit 44.1 kHz even if the hardware supports 24-bit 48 kHz.  Is this documented somewhere else?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a7582a35-f0e0-d7e5-b730-5925106afe45
* Version Independent ID: ee4e58e0-c9e5-265e-1c47-f064769abbbb
* Content: [Policy for Mixing Audio Streams and Setting the Output Sample Rate](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/policy-for-mixing-audio-streams-and-setting-the-output-sample-rate)
* Content Source: [windows-driver-docs-pr/audio/policy-for-mixing-audio-streams-and-setting-the-output-sample-rate.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/policy-for-mixing-audio-streams-and-setting-the-output-sample-rate.md)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **windowsdriverdev**

endolith, 2018-03-19:
Actually, now that I test more carefully, this is different in Windows 10 vs previous versions, but still not consistent.  

If a device supports

- Playback
  - 2 ch, 16-bit
    - 8000, 11025, 16000, 22050, 32000, 44100, 48000
- Recording
  - 1 ch, 16-bit
    - 8000, 11025, 16000, 22050, 32000, 44100, 48000

Then
- Windows 7 and 8.1 choose 44100 for both input and output
- Windows 10 chooses 44100 for input and 48000 for output

If a device supports

- Playback
  - 2 ch, 16-bit
    - 32000, 44100, 48000
  - 2 ch, 24-bit
    - 44100, 48000
- Recording
  - 2 ch, 24-bit
    - 44100, 48000

Then
- Windows 7 and 8.1 choose 48000 for input and 44100 for output
- Windows 10 chooses 48000 for both input and output

And a device that supports 96 kHz still defaults to 44.1 or 48 kHz.

endolith, 2018-03-20:
Related, but maybe for a separate Issue:

- If it supports 1 channel in, 16 | 24-bit, 32 | 44.1 | 48 kHz, Windows 7 and 8.1 will not show the 24-bit options at all.  
- If it supports 1 channel in, 24-bit, 32 | 44.1 | 48 kHz, Windows 7 and 8.1 will only show the 48 kHz option and not allow anything else to be selected.
- No Windows version ever displays 24-bit 32 kHz as an option even if it's supported (which is reasonable, but I can't find any documentation of it)

I've never found any documentation for how the different versions handle these things.  It's difficult to know if Windows just doesn't support something, or if there's a bug in my descriptors.

DOMARS, 2018-03-27:
I’m checking with the dev team to see if it is possible to provide this information. Thanks for your interest in improving these pages.

DOMARS, 2018-08-17:
@endolith - Sorry for the long delay. The audio team is interested in learning more about the sample rate testing that you did in the entry above labeled "Related, but maybe for a separate Issue:".  Could you provide a Windows build version and the audio devices that are used in each case?

endolith, 2018-08-18:
@DOMARS I develop USB audio devices at work, and tried altering the USB descriptors to those different configurations to see what Windows would do with them.  I can get the Windows build numbers in a few days.

(Ideally, there would be a way to prevent Windows from requesting different sample rates for input and output, but it seems Windows 10 does it by default now.  MacOS never requests different sample rates, for comparison.)

endolith, 2018-08-20:
- Windows 7 Professional
  - Version 6.1 (Build 7601: Service Pack 1)
- Windows 10 
  - Version 1803 (OS Build 17134.228)

DOMARS, 2018-09-14:
@endolith - Thanks for the build numbers. The dev team is looking at this issue and was wondering if the hardware that you develop is publically available? They may be able to replicate with out the specific hardware, but any addtional information that you can publically share would be helpful, if that is possible.

DOMARS, 2018-10-29:
@endolith - I followed up in email.

DOMARS, 2019-06-05:
@endolith - As the dev team is following up on this and has an internal tracking item, closing this issue.

closed 2019-06-05 by DOMARS