MicrosoftDocs/windows-driver-docs #377: Question about Driver Requirements

armstrong-charles, 2018-03-19:
If a driver package includes an *.exe, zip, *.msi, or *.cab that extracts signed binaries, is it ok? (assume that driver package satisfies other requirements too) 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 138a9602-43a6-1b8a-7159-dc3cd7e62f5a
* Version Independent ID: dbf5e2e6-51e1-a43b-becb-3919dc51cde3
* Content: [Windows 10 S Driver Requirements](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/windows10sdriverrequirements)
* Content Source: [windows-driver-docs-pr/install/Windows10SDriverRequirements.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/Windows10SDriverRequirements.md)
* Service: **unspecified**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-03-21:
Sorry, @armstrong-charles but the Dev Center can't accept compressed/archived binaries.  This prevents application of the WHQL signature

closed 2018-03-21 by tedhudek

lixiaogangkkk, 2019-01-08:
Can I install my customer DCHU driver by executing INF for Windows 10 S? Or Windows Update is the only way to update device driver for Windows 10 S?

Thanks,
Steven