MicrosoftDocs/windows-driver-docs #378: Missing error descriptions

keysightedwarda, 2018-03-19:
This article contains only a small subset of errors from InfVerif, and there are no other resources for InfVerif errors.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 48275f87-4086-1ca5-a190-c741f29ad348
* Version Independent ID: 5481885d-3063-527b-a85e-6a898dbcb990
* Content: [INF Validation Errors and Warnings](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/inf-validation-errors-and-warnings)
* Content Source: [windows-driver-docs-pr/devtest/inf-validation-errors-and-warnings.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/inf-validation-errors-and-warnings.md)
* Service: **unspecified**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-03-21:
Thanks for the suggestion @keysightedwarda are there specific error codes you're curious about?  There's nothing secret about these, but our existing resources don't allow us to craft an individual description for each one.  Happy to try and help if there are a few that are puzzling you.

keysightedwarda, 2018-03-21:
In a AddProperty section for a ClassInstall32 section, I'm getting the following error:
ERROR 1081: AddProperty directive has invalid property name "DeviceIcon".

"DeviceIcon" appears as a valid property in https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-addproperty-directive

zlockard, 2018-04-02:
@keysightedwarda This error indicates that we couldn't identify the property you were trying to set.  DeviceIcon is a well-known property which I was able to set in my testing, so I suspect that there might be something else going on.  Can you share more of the INF?  At least the AddProperty section would be a good start

keysightedwarda, 2018-04-03:
Here's the  ClassInstall32 section and it's child sections:

`[ClassInstall32]`
`AddReg = AgClassAddReg`
`AddProperty = AgClassAddProperty`

`[AgClassAddReg]`
`HKR,,,,%AgClassName%`
`HKR,,EnumPropPages32,,"AgM9290AResources.dll"`
`HKR,,Icon,,"101"`

`[AgClassAddProperty]`
`DeviceIcon,,,,"AgM9290AResources.dll,-102"`

Now that I look at this, it seems the team that produced this may be setting the class icon twice. Commenting out the AddReg directive and its section does not fix my issue, though.

What's more, InfVerif doesn't have this problem with the following section, referenced by the AddProperty directive in the DDInstall section:

`[AgAddProperty]`
`DeviceIcon,,,,"AgM9290AResources.dll,-102"`

Thanks,
-Ed

zlockard, 2018-04-03:
Sorry I missed this earlier, DeviceIcon is a device property, and can only be set via a DDInstall.  I don't believe there's a way to set the DeviceIcon for all devices that are a member of the class, just the class icon.

keysightedwarda, 2018-04-03:
I believe that the intention is to set the device class icon. I'd fix the property name, but I can't seem to find documentation for what that property is.

zlockard, 2018-04-03:
It's being set here:  HKR,,Icon,,"101"
You can also use the property syntax you were using before, where it's just "Icon"

closed 2018-04-17 by tedhudek