MicrosoftDocs/windows-driver-docs #380: Dead link

endolith, 2018-03-20:
Link to "[Audio Device Technologies for Windows](https://www.microsoft.com/whdc/device/audio/default.mspx)" is dead.  "We are sorry, the page you requested cannot be found."

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 31d404be-d32a-fa94-3670-3586230b730c
* Version Independent ID: 5d7e6fef-e833-46c3-07f0-fc4187d7c48a
* Content: [USB device class drivers included in Windows](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/supported-usb-classes)
* Content Source: [windows-driver-docs-pr/usbcon/supported-usb-classes.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/supported-usb-classes.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

endolith, 2018-03-20:
as are links to 

https://www.microsoft.com/whdc/device/input/default.mspx
https://www.microsoft.com/whdc/device/stillimage/default.mspx
https://www.microsoft.com/whdc/device/print/default.mspx
https://www.microsoft.com/whdc/device/storage/default.mspx
etc

DOMARS, 2018-03-27:
The broken links have been repaired. @endolith, thanks for pointing this out.

closed 2018-03-27 by DOMARS