MicrosoftDocs/windows-driver-docs #381: How to prevent Windows from requesting conflicting sample rates?

endolith, 2018-03-20:
> hardware typically has a single crystal, which constrains its input and output streams to run at the same clock rate

Is there any way to indicate this constraint to Windows for USB Audio devices?  [Different versions of Windows have different rules for default sample rates](https://github.com/MicrosoftDocs/windows-driver-docs/issues/376#issuecomment-374393318), which leads to conflicting input/output rates and garbled audio by default.

If conflicting rates are requested, we can STALL the second request, as per USB spec:

> If an audio function does not support a certain request, it must
indicate this by stalling the control pipe when that request is issued to the function.

but then recording software locks up or behaves unusually, and Windows doesn't try again with a compatible rate or tell the user what the problem is.

USB Audio 2.0 descriptors support clock information, which might prevent Windows from requesting conflicting rates, but Windows doesn't fully support UAC 2.0 yet.

(OS X etc. never request conflicting in/out rates, so never have this problem.)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1d5ac36b-6b5b-030b-6e83-b6e9155f0ce5
* Version Independent ID: 7d8b6999-0f1a-d5c1-a119-9e43f9f53399
* Content: [Hardware Constraints on Sample Frequency](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/hardware-constraints-on-sample-frequency)
* Content Source: [windows-driver-docs-pr/audio/hardware-constraints-on-sample-frequency.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/hardware-constraints-on-sample-frequency.md)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **windowsdriverdev**

DOMARS, 2018-08-06:
@endolith - Following up with dev team to see if I can get any information on this request. They have been heads down coding.

DOMARS, 2018-09-14:
@endolith - The dev team is investigating this code behavior, that work will take some additional time.

DOMARS, 2019-06-05:
@endolith - As the dev team is following up on this and has an internal tracking item, closing this issue.

closed 2019-06-05 by DOMARS

endolith, 2019-06-17:
(This was fixed for the default sample rate (#376), but it's still possible to manually select conflicting sample rates.  Hopefully, anyone who finds where to do this also realizes they need to change both input and output.)