MicrosoftDocs/windows-driver-docs #382: Similar document for USB Audio 1.0?

endolith, 2018-03-20:
Is there a similar document that lists the capabilities of the USB Audio 1.0 driver?  A table of supported formats, restrictions on topology, table of supported control requests, "does not support implicit feedback", etc.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7bc8e063-bf45-422a-5918-b5dafc707542
* Version Independent ID: af0056d7-cc8f-0e0b-e252-9a149b9a2530
* Content: [USB Audio 2.0 Drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/usb-2-0-audio-drivers)
* Content Source: [windows-driver-docs-pr/audio/usb-2-0-audio-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/usb-2-0-audio-drivers.md)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **windowsdriverdev**

DOMARS, 2018-03-27:
I’m checking with the dev team to see if it is possible to provide this information. Thanks for your interest in improving these pages.

lixiaogangkkk, 2018-07-05:
Dear  MS guy,
     I have two question about this USB Audio 2.0 class device driver.
    1. Does it support audio jack detection feature as HD Audio device? (device status changes to "not plugged in" status while analog microphone/headphone is un-plugged from device).
    2. Could you share any experience about how does audio buffer updated between Audio Engine and Audio driver, Between Audio driver and USB ISO pipe under WaveRT Pull Mode (Event Driver Mode）. Now I'm working on supporting WaveRT Pull Mode in my USB audio driver, but I'm not clear how to update the buffer and always get "pop" noiese (data drop).

Thanks,
Synaptics Inc, Steven Li

DOMARS, 2018-08-06:
@endolith - Following up with dev team to see if I can get any information on this request. They have been heads down coding.

DOMARS, 2018-08-06:
@lixiaogangkkk  - There is some general information on Wave RT latency in this topic.
https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/stream-latency-during-playback

On you other questions, unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.
You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

- Windows Hardware WDK and Driver Development
https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk

- OSR's ntdev List
https://www.osronline.com/showlists.cfm?list=ntdev

Best of luck with your work.

DOMARS, 2018-09-14:
@endolith - Like issue 383, our dev team has created an internal work item to investigate creating this content for UAC1. This request will of course need to be balanced against other feature/bug work. Having this information documented would be useful, so thanks for the request and thank you for the interest in improving our audio driver documentation

lixiaogangkkk, 2018-11-22:
Question：
1）Is there a plan to support LPM with USB Audio driver? I checked that Google has supported UAC2.0 with LPM after Android 8.1 original system. Pixel 2 and Pixel 3 are shipped out. 
2)   Is there a plan to support UAC3.0? I heard that latest Linux kernel (4.19, Ubuntu) already supported UAC3.0.

Thanks,
Steven

DOMARS, 2019-06-05:
@lixiaogangkkk - On your other questions, unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.
You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.
Windows Hardware WDK and Driver Development
https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk
OSR's ntdev List
https://www.osronline.com/showlists.cfm?list=ntdev
Best of luck with your work.

closed 2019-06-05 by DOMARS