MicrosoftDocs/windows-driver-docs #383: More information about capabilities and restrictions

endolith, 2018-03-20:
Can this be expanded to include more information about restrictions on topology, table of supported control requests, supported isochronous types, implicit feedback, etc.?  Similar to https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/usb-2-0-audio-drivers  https://github.com/MicrosoftDocs/windows-driver-docs/issues/382

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: eed3af2f-9a8c-7baf-53f2-d57c71d00882
* Version Independent ID: cb50b482-f74f-c28b-4186-ebd8e26e398f
* Content: [USB Audio Class System Driver (Usbaudio.sys)](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/usb-audio-class-system-driver--usbaudio-sys-)
* Content Source: [windows-driver-docs-pr/audio/usb-audio-class-system-driver--usbaudio-sys-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/usb-audio-class-system-driver--usbaudio-sys-.md)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **windowsdriverdev**

DOMARS, 2018-03-27:
I’m checking with the dev team to see if it is possible to provide this information. Thanks for your interest in improving these pages.

DOMARS, 2018-08-06:
@endolith - Following up with dev team to see if I can get any information on this request. They have been heads down coding.

DOMARS, 2018-09-14:
@endolith - Our dev team has created an internal work item to investigate creating this content for UAC1. This request will of course need to be balanced against other feature/bug work. Having this information documented would be useful, so thanks for the request and thank you for the interest in improving our audio driver documentation

DOMARS, 2019-06-05:
@endolith - As the dev team is following up on this and has an internal tracking item, closing this issue.

closed 2019-06-05 by DOMARS