MicrosoftDocs/windows-driver-docs #385: missing pic

maperr, 2018-03-20:
Picture is missing for #9

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 81568a12-c39e-a18f-8f33-c54545f14bac
* Version Independent ID: 73217b22-1795-b5d0-65cb-1984f38dc661
* Content: [Write a universal Hello World driver (KMDF)](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/writing-a-very-small-kmdf--driver)
* Content Source: [windows-driver-docs-pr/gettingstarted/writing-a-very-small-kmdf--driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/writing-a-very-small-kmdf--driver.md)
* Service: **unspecified**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-03-21:
Thanks, @maperr! Just a couple of extra blank lines causing mischief. I've made the change and merged to the staging branch. Should be live by the morning.

EliotSeattle, 2018-03-21:
Merged to staging

closed 2018-03-21 by EliotSeattle