MicrosoftDocs/windows-driver-docs #387: Issue with Example of a kernel-mode driver using WPP_CLEANUP

ayodejiige, 2018-03-22:
What it contains:
    WPP_CLEANUP(WdfDriverWdmGetDriverObject(DriverObject));
What I think it should be:
    WPP_CLEANUP(DriverObject);

Is DriverObject not already a PDRIVER_OBJECT?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f5d3d49a-28cd-231f-d58e-743cd6bc02f5
* Version Independent ID: 7adf2d54-13be-6458-77d6-280de3464cff
* Content: [Adding WPP Software Tracing to a Windows Driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/adding-wpp-software-tracing-to-a-windows-driver)
* Content Source: [windows-driver-docs-pr/devtest/adding-wpp-software-tracing-to-a-windows-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/adding-wpp-software-tracing-to-a-windows-driver.md)
* Service: **unspecified**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

duncanmacmichael, 2018-03-22:
Hi @ayodejiige , thanks for pointing out this issue. Technically, the WPP_CLEANUP call is correct because that macro requires a WDM driver object. The root of the confusion is that the previous DriverEntry example worked with a WDF-based driver (as it called WdfDriverCreate), but the context cleanup example doesn't make it clear that this is a cleanup for a WDF driver object. So, the driver context cleanup should have a WDFDRIVER as an input parameter instead of a WDM PDRIVER_OBJECT. More accurately, that cleanup example should be re-named to EVT_WDF_DRIVER_UNLOAD. I'll fix that for you so it's clearer why WPP_CLEANUP is specifying the WDM driver object that is associated with the WDFDRIVER object passed in.

Does that help?

duncanmacmichael, 2018-03-22:
I've updated the example for you to hopefully make this clearer. The changes should be live soon. Thanks again for the sharp eye.

closed 2018-03-22 by duncanmacmichael

kouzhudong, 2019-03-26:
为何在DriverUnload里调用了WPP_CLEANUP，还是会出现：
DRIVER_VERIFIER_DETECTED_VIOLATION (c4)
A device driver attempting to corrupt the system has been caught.  This is
because the driver was specified in the registry as being suspect (by the
administrator) and the kernel has enabled substantial checking of this driver.
If the driver attempts to corrupt the system, bugchecks 0xC4, 0xC1 and 0xA will
be among the most commonly seen crashes.
Arguments:
Arg1: 00000000000000dd, Unloading driver that forgot to call EtwUnregister.
Arg2: fffff880046a76d0, Address where the culprit driver called EtwRegister.
Arg3: fffff88004665000, Start address of the culprit driver.
Arg4: 0000000000000000, Address of the leaked ETW_REG_ENTRY structure.
我怀疑WPP_CLEANUP的代码有问题，或者Windows的机制有问题，如：verifier。

kouzhudong, 2019-03-26:
@duncanmacmichael

duncanmacmichael, 2019-03-26:
Hi @kouzhudong, 

I'm sorry you're having an issue but unfortunately we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.

You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

• [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
• [NTDEV forum at OSR Online](https://community.osr.com/)

Again, thanks for reaching out. Best of luck with this issue.

kouzhudong, 2019-06-19:
问题已经提交到：https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xc4--driver-verifier-detected-violation