MicrosoftDocs/windows-driver-docs #389: No writing takes place whatsoever

Shorrer, 2018-03-24:
Hello there.
Short story: I came here to get some knowledge about writing drivers for Windows. Prerequisites for this "Getting started with Windows drivers" was C/C++ knowledge and, obviously, some experience with actual programming, because you, obviously, can't write drivers if you don't understand a thing about programming. So, I've expected this "Write your first driver" to be actual tutorial on how to *write* your first driver, and not "how to create project and test if you installed WDK correctly". I know how to use Visual Studio, and I think that many other people who came here also know this (if not, then you can just send them down the link to "how to create projects in our Visual Studio" and be done with it). What I don't know is how drivers work and how to code them properly, and this article says absolutely and completely zero bits of information about that.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 81568a12-c39e-a18f-8f33-c54545f14bac
* Version Independent ID: 73217b22-1795-b5d0-65cb-1984f38dc661
* Content: [Write a universal Hello World driver (KMDF)](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/writing-a-very-small-kmdf--driver)
* Content Source: [windows-driver-docs-pr/gettingstarted/writing-a-very-small-kmdf--driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/writing-a-very-small-kmdf--driver.md)
* Service: **unspecified**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-03-25:
Thanks for the feedback, @Shorrer.

I'll take a look at how we can either rename these topics to be more transparently correct and to point to what it really means to write your first driver, which may mean writing a book...

Thank you so much for taking the tiome to respond.

Shorrer, 2018-03-28:
Maybe these can be moved from "Programming guides" section (I refer to sections at this page https://docs.microsoft.com/en-us/windows-hardware/drivers/index) to "Code samples". Actual name can be something like "Compile your first drive", so people wouldn't expect from them a full-blown explanation on how to write your own drivers. It is really a problem of slightly deceiving naming because, as you noted and as searching through the web proved, it really takes a book to explain writing drivers.
Thanks for the response.

tedhudek, 2018-04-18:
Thanks for the feedback, @shorrer.  I'm just at a bit of a loss what to do with it.

The goal of this topic is to provide everything you need to create Hello World, from creating the VS project, to adding the minimal code, to building and deploying the driver, with a few hints on debugging.  These are all necessary steps when writing a driver.

I suppose we could change the word "write" to "create" in the title, but I'm not convinced that would add any value.

Also note that the "Programming guides" section is just a set of links to useful pages to start your driver journey.  I think this page belongs there much more than in Code Samples, which is the link to a separate driver sample repo.

closed 2018-04-18 by tedhudek

duncanmacmichael, 2018-04-20:
Hi @Shorrer , we've gone back and added some more explanation to the "writing code" step. While there is still a certain amount we need to prescribe for the first driver code you write, we broke it down into a few more steps and explained what you were writing along the way with comments. Do you think this is more accessible to programmers who are just starting with writing drivers?

Shorrer, 2018-04-25:
Yeah, this suits the title exactly the right way: now it is perfectly clear what is used for what and how driver should be structured at all. Links to programming reference for functions and objects are also there, so it is now possible to easily get more specific information if it's required.
I want to thank everyone involved in the edits that was made to this page. This page has gone from "copypaste this into your editor" to easy for understanding explanatory article about basic driver development principles, which is very useful.

duncanmacmichael, 2018-04-25:
You're welcome, happy to hear it! Thanks for helping us improve the docs. We're always looking for ways to make them better, so collaborative feedback like this is very valuable.