MicrosoftDocs/windows-driver-docs #393: Broken link

GabrieleGiuseppini, 2018-03-26:
The link to the Windows Symbols Web site for "XP or later" is broken.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 81766d8e-917c-bc0a-a210-14c373c36afe
* Version Independent ID: 88c6b1d8-063c-8e14-3b78-37469b541012
* Content: [Installing Windows Symbol Files](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/installing-windows-symbol-files#feedback)
* Content Source: [windows-driver-docs-pr/debugger/installing-windows-symbol-files.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/installing-windows-symbol-files.md)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-03-26:
Thank you, @GabrieleGiuseppini. We'll get on that today.

DOMARS, 2018-03-27:
@GabrieleGiuseppini, I updated this page and fixed the broken link. Thanks for pointing this out.

closed 2018-03-27 by DOMARS