MicrosoftDocs/windows-driver-docs #397: Need info on how to find the "legal agreements" mentioned on this page.

fpl9000, 2018-03-27:
A bullet on this page says "Sign the required hardware developer legal agreements", but there is no info provided on where to find those legal agreements. I think this page should either include that info or provide a link to a page that provides that info. Thanks.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2daf70ed-7391-5783-583b-fcd99f1c222a
* Version Independent ID: 679bd45d-d0fd-a192-4d26-0121594fa86a
* Content: [Register for the Hardware Program](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/register-for-the-hardware-program)
* Content Source: [windows-driver-docs-pr/dashboard/register-for-the-hardware-program.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/register-for-the-hardware-program.md)
* Service: **unspecified**
* GitHub Login: @aahill
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-03-27:
Thanks, @fpl9000 for the feedback. @aahill, can you make the statement clearer that the legal agreements are in the tool?

aahill, 2018-03-30:
Hi @fpl9000, I've just updated the documentation to clarify where you can find the required legal agreements, with screenshots. You can see it on the page you accessed: https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/register-for-the-hardware-program

Hopefully this makes things less confusing. Thanks for reaching out!

closed 2018-03-30 by aahill