MicrosoftDocs/windows-driver-docs #398: Needs clarification of UNICODE_STRING versus Unicode string, and link to du

randomascii, 2018-03-27:
A search (bing or Google) for "windbg print unicode string" will find this topic, rather than the du topic which is probably what is wanted. It takes a careful reading of the dS documentation to realize that "The memory address where the string begins." means "The memory address where the UNICODE_STRING object is stored."

And, even if a reader figures out that dS is the wrong function, they are still not closer to figuring out what the right function is (if they have the address of a Unicode string that they want to print).

The .chm version of the documentation has all of these failing but also has inaccessible du documentation, in order to add an extra barrier.

Requests:

- The sentence "The memory address where the string begins." needs to be rewritten as "The memory address where the UNICODE_STRING object is stored."
- There needs to be a link to the du command documentation
- Instead of just saying "These commands do not display null-delimited character strings." say "If you have the address of the characters of a Unicode string then use the du command instead."

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 33c9e19f-7b70-6ffe-97ae-6626756411f4
* Version Independent ID: 6757583f-a2cd-81e3-5add-c05c3c6bdede
* Content: [ds, dS (Display String)](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/ds--ds--display-string-)
* Content Source: [windows-driver-docs-pr/debugger/ds--ds--display-string-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/ds--ds--display-string-.md)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-04-11:
@randomascii - I did some testing today and I see what you mean. I should have an updated topic out pretty soon that highlights that ds dS is for string structures. Thanks for pointing this out.

DOMARS, 2018-04-13:
@randomascii - The updates to the ds, DS topic are live. Thanks again for your input on this topic.

closed 2018-04-13 by DOMARS