MicrosoftDocs/windows-driver-docs #399: Lacks of details on New DDIs on Windows RS2

saumyajyoti, 2018-03-28:
The new DDis like QueryConnectionChange, SetTimingsFromVidpn are not grouped in the DDI synchronization levels.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 518ba786-7bda-cf0c-4aa0-0dc08a1fe68c
* Version Independent ID: 32db11aa-26d4-b64c-c370-ad593b7f88b3
* Content: [Child I/O Class](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/child-i-o-class)
* Content Source: [windows-driver-docs-pr/display/child-i-o-class.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/child-i-o-class.md)
* Service: **unspecified**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

mimisasouvanh, 2018-04-02:
@saumyajyoti  Thank you for bringing this to our attention. I will confirm the right synchronization levels for the DDIs and insert them in the proper location.

mimisasouvanh, 2018-04-09:
@saumyajyoti  I've fixed the docs to include:

- DxgkDdiSettimingsFromVidPn, which is same as DxgkDdiCommitVidPn, in [Threading and Synchronization Second Level](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/threading-and-synchronization-second-level).
- DxgkDdiQueryConnectionChange, which is same as DxgkDdiQueryChildStatus, and DxgkDdiDisplayDetectControl, in [Child I/O Class](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/child-i-o-class).
- DxgkDdiSetTargetGamma, DxgkDdiSetTargetContentType, DxgkDdiSetTargetAnalogCopyProtection, and DxgkDdiSetTargetAdjustedColorimetry in a new class called [Display Class](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/display-class) at [Threading and Synchronization First Level](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/threading-and-synchronization-first-level).

Feel free to re-open the issue if there are other items missing.

closed 2018-04-09 by mimisasouvanh