MicrosoftDocs/windows-driver-docs #401: Forget include generated .rc file to project

hansel163, 2018-03-29:
You should mention include generated .rc file to visual studio project. Otherwise, it will report "resources could not be found or are not accessible" when install the manifest file.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2f93c1d9-ccc3-bc8a-14d9-7b23df2773c0
* Version Independent ID: 50326231-07f4-0f95-3dec-76fc49a75612
* Content: [Adding Event Tracing to Kernel-Mode Drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/adding-event-tracing-to-kernel-mode-drivers)
* Content Source: [windows-driver-docs-pr/devtest/adding-event-tracing-to-kernel-mode-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/adding-event-tracing-to-kernel-mode-drivers.md)
* Service: **unspecified**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-03-29:
Thanks, @hansel163. This should be live shortly.

matlimatli, 2023-11-01:
Now there is a note saying that you may get errors if the .rc file is not added. It would be more helpful to say *how* it should be added. One issue is for example that by following the instructions, the .rc file will be generated to different locations depending on build type (e.g. Debug/Release).