MicrosoftDocs/windows-driver-docs #408: Where is the example about 64bit windows for API MmGetPhysicalAddress

isdqwq, 2018-04-03:
Dear sir.
  how to write code to get the Physical Address on User-Space for 64BIT ioctl driver?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2f6e4cc7-700c-4969-802a-d73252424cf1
* Version Independent ID: 62a4529b-68e1-67c0-41e9-b918b559f1a1
* Content: [Performing DMA in 64-Bit Windows](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/performing-dma-in-64-bit-windows)
* Content Source: [windows-driver-docs-pr/kernel/performing-dma-in-64-bit-windows.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/performing-dma-in-64-bit-windows.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

PageWriter-MSFT, 2018-04-20:
Hello isdqwq,

Thanks for your request. After checking with the Windows Kernel team, such an operation seems unsafe because the physical address underlying any virtual address could change with no notification to the process.

Please post this question to a driver development forum (such as https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk) with more details and context.

Thanks

PageWriter

closed 2018-04-20 by PageWriter-MSFT

isdqwq, 2018-05-09:
thanks, i had fix this issue,it's work OK.