MicrosoftDocs/windows-driver-docs #409: AddSoftware  directive

MichaelSun123, 2018-04-03:
I used AddSoftware directive and assign the SoftwareType=1.
After I install the extension inf and then install the component inf. 
The binary I assigned by SoftwareBinary doesn't run. 
Then I try to use the DCHU example on GitHub. It still didn't work.
Does anyone can help explain how this directive work?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ffd84f19-a70c-f319-05d7-de54a24c90b7
* Version Independent ID: 08e48321-ba5b-6204-f795-b1e42cd093f6
* Content: [INF AddSoftware Directive](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-addsoftware-directive)
* Content Source: [windows-driver-docs-pr/install/inf-addsoftware-directive.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-addsoftware-directive.md)
* Service: **unspecified**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-04-03:
@NabilFates can you help?

NabilFates, 2018-04-19:
Which OS version are you using? Are you using the OSR board recommended in the DCHU sample?

MichaelSun123, 2018-04-23:
The OS Version is RS3. I don't have an OSR board. Actually, I just install the extension inf and component inf.  But I didn't see the legacy exe binary started to run. According to this document, it should run.  Did I miss any import information?

MichaelSun123, 2018-04-26:
Anybody can help ?

NabilFates, 2018-04-26:
Your extension driver needs to match on a device in order to create the virtual component device, and subsequently install the Component INF (which does the AddSoftware). 

If you don't have the OSR board, are you attempting to install your extension INF on another device?