MicrosoftDocs/windows-driver-docs #410: Getting rid of a linking error

lihas, 2018-04-03:
Also include initguid.h.
#include <initguid.h>.
Otherwise I was getting "unresolved external symbol _GUID_DEVINTERFACE_USB_DEVICE"

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dc07324f-bf4f-8857-2ea3-3539a586fb27
* Version Independent ID: 63179fd5-2bd5-ef2c-b180-419340116d29
* Content: [GUID_DEVINTERFACE_USB_DEVICE](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/guid-devinterface-usb-device)
* Content Source: [windows-driver-docs-pr/install/guid-devinterface-usb-device.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/guid-devinterface-usb-device.md)
* Service: **unspecified**
* GitHub Login: @tedhudek

tedhudek, 2018-04-18:
Thanks for the feedback, @lihas, are you suggesting that we add initguid.h to the Requirements section of this page?

PageWriter-MSFT, 2018-05-30:
Thanks for pointing that out. I have added the header to the requirements section.

closed 2018-05-30 by PageWriter-MSFT

gaborgsomogyi, 2020-12-19:
Oh man, saved me a lot of headaches :)