MicrosoftDocs/windows-driver-docs #412: Add a "Resolution" section

LuigiBruno, 2018-04-03:
Missing a troubleshooting and resolution list of steps.
One way to try fixing this problem is using the System Recovery options and execute the "chkdsk /r" command to check the integrity of the boot volume.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d4d2aa14-6256-865d-b5c8-0e5c69c7224a
* Version Independent ID: c595bf21-ca5a-d8e0-560d-215f9169fa17
* Content: [Bug Check 0xED UNMOUNTABLE_BOOT_VOLUME](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xed--unmountable-boot-volume)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0xed--unmountable-boot-volume.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0xed--unmountable-boot-volume.md)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-06-25:
Planning to consolidate the debugging issues and get with the dev team this week.

DOMARS, 2018-06-27:
@LuigiBruno - I added a resolution section that includes chkdsk /r.The updated topic is live. Thanks for this suggestion.

closed 2018-06-27 by DOMARS