MicrosoftDocs/windows-driver-docs #415: ServiceName appears to be limited in length to 31 characters.

wes-b, 2018-04-05:
ServiceName appears to be limited in length to 31 characters. Anything longer and the device will fail installation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6bd0d4c0-7cfd-278b-12e4-a6532269db54
* Version Independent ID: 660209eb-539f-6ad5-dc93-7879d84308a1
* Content: [Specifying WDF Directives in INF Files](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/specifying-wdf-directives-in-inf-files)
* Content Source: [windows-driver-docs-pr/wdf/specifying-wdf-directives-in-inf-files.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/wdf/specifying-wdf-directives-in-inf-files.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-04-19:
Hi @wes-b, thanks for the feedback.  I took a brief look at the page and don't see where you're suggesting adding this info.  Maybe submit a PR with your proposed change?

wm1, 2018-04-25:
The max length is defined in public WdfGlobals.h file:
   `#define WDF_DRIVER_GLOBALS_NAME_LEN (32)`

tedhudek, 2018-05-29:
Added this to the page, will be live later today. Thanks for the heads-up @wes-b

closed 2018-05-29 by tedhudek