MicrosoftDocs/windows-driver-docs #416: Broken link to ACPIGen folder

rohitdubey12, 2018-04-05:
The link to ACPIGen folder directs to the Windows-driver-samples repository on github. Where can I find the ACPIGenFx.dll file?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 16299ebf-80ed-f09f-d0c0-4cda742f579e
* Version Independent ID: 8e45d880-93f8-4c85-81e6-b1d5dbe78ab6
* Content: [Generate ACPI tables by using AcpiGenFx](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/generate-acpi-tables-by-using-acpigenfx#feedback)
* Content Source: [windows-driver-docs-pr/bringup/generate-acpi-tables-by-using-acpigenfx.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/generate-acpi-tables-by-using-acpigenfx.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

barrygolden, 2018-04-05:
@rohitdubey12, thanks for pointing out this broken link.

AcpiGenFx and associated samples are available in the **Tools** folder of the WDK.
 
In the **Tools** directory, navigate to the target architecture folder, then to the **AcpiGenFx** folder.

For example, the x86 version is located in **C:\Program Files (x86)\Windows Kits\10\Tools\x86\ACPIGenFx**.

I'll update the topic with this information.

barrygolden, 2018-06-26:
Added information to the topic on the location of AcpiGenFx and associated samples in the WDK Tools directory.

closed 2018-06-26 by barrygolden